OpenTabletDriver/OpenTabletDriver #1299: Gaomon s56k not being detected

EikelFlo, 2021-07-30:
## What I've already tried
WinUSBCleanup

## Diagnostic Information
`{
  "App Version": "OpenTabletDriver v0.5.3.3",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.22000.0",
    "VersionString": "Microsoft Windows NT 10.0.22000.0"
  },
"HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_046d&pid_c547&mi_02&col01#8&282242b6&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 1133,
      "ProductID": 50503,
      "CanOpen": true,
      "ReleaseNumberBcd": 1024,
      "ReleaseNumber": "4.0",
      "ProductVersion": 1024,
      "Manufacturer": "Logitech",
      "ProductName": "USB Receiver",
      "SerialNumber": "",
      "MaxInputReportLength": 7,
      "MaxOutputReportLength": 7,
      "MaxFeatureReportLength": 0
    },
    {
      "DevicePath": "\\\\?\\hid#vid_046d&pid_c547&mi_01&col03#8&44b04f4&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 1133,
      "ProductID": 50503,
      "CanOpen": true,
      "ReleaseNumberBcd": 1024,
      "ReleaseNumber": "4.0",
      "ProductVersion": 1024,
      "Manufacturer": "Logitech",
      "ProductName": "USB Receiver",
      "SerialNumber": "",
      "MaxInputReportLength": 2,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 0
    },
    {
      "DevicePath": "\\\\?\\hid#vid_05ac&pid_024f&mi_00#8&192e7158&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd",
      "VendorID": 1452,
      "ProductID": 591,
      "CanOpen": false,
      "ReleaseNumberBcd": 260,
      "ReleaseNumber": "1.4",
      "ProductVersion": 260,
      "Manufacturer": "Keychron",
      "ProductName": "Keychron K2",
      "SerialNumber": "",
      "MaxInputReportLength": 9,
      "MaxOutputReportLength": 2,
      "MaxFeatureReportLength": 65
    },
    {
      "DevicePath": "\\\\?\\hid#vid_19f7&pid_0003&mi_03&col01#8&1b75e848&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 6647,
      "ProductID": 3,
      "CanOpen": true,
      "ReleaseNumberBcd": 256,
      "ReleaseNumber": "1.0",
      "ProductVersion": 256,
      "Manufacturer": "R",
      "ProductName": "ROD",
      "SerialNumber": "",
      "MaxInputReportLength": 16,
[... 1122 lines cut ...]
      "Time": "2021-07-30T16:16:15.1000318+02:00",
      "Group": "Settings",
      "Message": "Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ], [6, ], [7, ]",
      "Level": 1
    },
    {
      "Time": "2021-07-30T16:16:15.100042+02:00",
      "Group": "Settings",
      "Message": "Driver is auto-enabled.",
      "Level": 1
    },
    {
      "Time": "2021-07-30T16:16:15.1000494+02:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "Level": 0
    },
    {
      "Time": "2021-07-30T16:16:15.1000567+02:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "Level": 0
    }
  ]
}`

Mrcubix, 2021-07-30:
this S56K variation will only work on linux and macOS until linux is implemented, duplicate of  
Duplicate of #1068

closed 2021-07-30 by Mrcubix

EikelFlo, 2021-07-30:
> this S56K variation will only work on linux and macOS until WinUSB is implemented, duplicate of
> Duplicate of #1068

> > Yeah, I have the same issue and my dump says Huion Tablet LuoYong70558 will just wait until its solved whenever the backlog is cleared
> 
> Open your own issue, this probably has nothing to do with your

Mrcubix, 2021-07-30:
There has been working variations of the S56K, your isn't, that's it