OpenTabletDriver/OpenTabletDriver #1300: Add support for Acepen AP960

FlyMyPG, 2021-07-30:
# Tablet Information

## Base Information
| Name            | Value |
| --------------- | ----- |
| Manufacturer | Acepen |
| Product Name | AP906 |
| Drawing Area Width, Inches | 9 |
| Drawing Area Height, Inches | 5 |
| Pressure Levels | 8192 |
| Read Resolution, LPI | 5080 |
| Reading Speed, Points/Sec | 266 |
| Tilt Range Degrees | +/- 60 |

## Extra Information
| Name            | Value |
| --------------- | ----- |
| Mfr. URL | https://www.ace-pen.com
| Product URL | https://www.ace-pen.com/graphic-tablet/animation-digital-drawing-tablet/battery-free-animation-design-digital-drawing.html
| Driver URL | https://www.ace-pen.com/support
| Amazon URL | https://www.amazon.com/gp/product/B07VYNLMV8
| Linux work | https://githubmemory.com/repo/DIGImend/digimend-kernel-drivers/issues/540

## Diagnostic Information
Diagnostics soon: Need to get some work done with existing drivers.

## USB Device Descriptor
    DEVICE DESCRIPTOR
        bLength: 18
        bDescriptorType: 0x01 (DEVICE)
        bcdUSB: 0x0110
        bDeviceClass: Device (0x00)
        bDeviceSubClass: 0
        bDeviceProtocol: 0 (Use class code info from Interface Descriptors)
        bMaxPacketSize0: 64
        idVendor: UC-Logic Technology Corp. (0x5543)
        idProduct: Unknown (0x0091)
        bcdDevice: 0x0000
        iManufacturer: 1
        iProduct: 2
        iSerialNumber: 3
        bNumConfigurations: 1

## Wireshark Captures of Tablet Functions
[WireShark.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6908484/WireShark.zip)

Mrcubix, 2021-07-30:
Still awaiting a diagnostic that literally take 2 click to get (in OTD: Help > export diagnostic)

FlyMyPG, 2021-07-30:
No, you are wrong.  I'd have to uninstall the OEM driver to avoid a conflict with OTD.  It took me a while to get the OEM driver working well enough to use.  My work is progressing.

I'm not installing OTD until my current work is done.  I can't afford the risk of things going wrong.

I'm also working with Acepen to resolve the worst issues with their driver and user-space tool.  I joined their tester program to get pre-release drops.  However, my hopes are not high this will go anywhere, which is why I looked at this project in the first place.

I'm juggling lots of balls here.  Please be patient.

Kuuuube, 2021-07-30:
There will be zero progress on this until you are willing to help. There is no reason to create an issue in advance and doing so makes it more likely your issue will never get attention when you do decide to provide the required information and can test configs.

I'm closing this issue here. Feel free to create another issue when you are more available.

closed 2021-07-30 by Kuuuube

FlyMyPG, 2021-07-30:
And I'm done with this project.  I supplied lots of information you apparently rejected.

I opened the issue to keep it on my github list, so I will return to it right after my current work is done.  But that's not good enough for you.

Whatever.  I hope you love your life.

Mrcubix, 2021-07-31:
Thing is we know when huion manufactured tablet will be compatible or not, you apparently don't, we ask for information, and you act like a god saying you know more than us, stop acting like a child and make a proper issue with only an OTD diagnostic, it will literally take 5s to know if it's compatible or not with one, no matter if OEM drivers are present or not.

FlyMyPG, 2021-07-31:
You are the child who has no patience.  I said I would provide the dump right after my current work completed, which should be 3-4 days (hopefully).

I provided lots of other information.  Did you bother to look at it?  Probably not.  You just want what you want, and don't care about the rest of the world.

So be it.  It is your project, I respect that, though I choose not to participate according to your demands.

I need to use my current driver for a little longer.  Too bad that doesn't work for you.  I don't care.  Bye!

doubledge-spec, 2021-07-31:
You provided wireshark information. I assume that you were concerned with OTD interfering with anything on your system. What should have been the answer was,,,, OTD is a fully user mode driver. It will not interfere with any sensitive application. Both sides handled the situation sub optimally. Hopefully in the future you will have more positive interactions.

Mrcubix, 2021-07-31:
the only information we need to make a config here would be the name of the tablet, the diagnostic file which 95% of the time contain the information needed to identify if this huion manufactured tablet is compatible (the 5 remaining % being UGEE manufactured tablets) and the wireshark, tho i don't know if it was captured using USBPcap.

doubledge-spec, 2021-07-31:
I didn't really pay attention until now, also I have horrible memory retention. I only remembered when I looked that this thread now. My apologies.

FlyMyPG, 2021-07-31:
Closing the ticket immediately, not even letting it sit for a few days, was a total jerk move.  It's your project to run however you like.

I saw only that OTD installation was required, and I did NOT want to install another tablet driver.  If your documentation sucks, that's on you.  It didn't work this time around, that's for sure.

Mrcubix, 2021-07-31:
told him on discord, he knows and he said he don't like dealing with github support, but we learn.

Also OTD doesn't install anything, it's a portable application, and the installer just download the portable version and store it in %localappdata%/opentabletdriver, it doesn't run in background, doesn't install devices, or anything.

Anyway i can re-open that issue since we aren't done yet, unless you give up.

FlyMyPG, 2021-07-31:
No, I'm just venting here.  I have no future interest in OTD so long as the project operates like it did for this ticket.

Mrcubix, 2021-07-31:
Alright that's your choice, just know that i wouldn't had closed that myself.
Also nothing prevent you from coming to the [discord server](https://discord.gg/P3GYdUS) where we'll be able to guide you to export the necessary diagnostic in 5 minutes without it disturbing your other driver.

FlyMyPG, 2021-07-31:
Put that in the HowTo for "Requesting A New Tablet To Be Supported".

Mrcubix, 2021-07-31:
I would agree on changing some part i'll let The ghost know

FlyMyPG, 2021-07-31:
That said, I do not want @Mrcubix to interact on any future ticket I post (assuming that ever happens).  Toxic.

InfinityGhost, 2021-07-31:
This entire ticket is a mess, I'm locking this until further notice.

If you'd wish to start anew with a fresh ticket without drama and aimless arguments the contribution is welcome.