OpenTabletDriver/OpenTabletDriver #1301: gaomon m10k pro is not detected

no6hope, 2021-07-30:
## Description

## System Information:

| Name             | Value |
| ---------------- | ----- |
| S.O win 10 pro 64bits |
| Software Version https://github.com/OpenTabletDriver/OpenTabletDriver.Installer/releases/tag/v0.4.1 |
| Tablet     gaomon m10k pro      |

MegaHello, 2021-08-04:
Have you installed the Gaomon Drivers before?

no6hope, 2021-08-05:
I've installed it on my cousin's computer, and it works (gaomon driver) ,
but I've not installed the gaomon driver on my pc

Mrcubix, 2021-08-05:
fill the template by providing a diagnostic file from OTD

no6hope, 2021-08-06:
Is it? De: Gess1tEnviado: jueves, 5 de agosto de 2021 12:30Para: OpenTabletDriver/OpenTabletDriverCC: no6hope; AuthorAsunto: Re: [OpenTabletDriver/OpenTabletDriver] gaomon m10k pro is not detected (#1301) fill the template by providing a diagnostic file from OTD

MegaHello, 2021-08-06:
Please don't use email to reply to the issue. It's gonna look really weird on our side.

no6hope, 2021-08-07:
## Description

## System Information:

| Name             | Value |
| no6hope | ?|
| Operating System | OS Windows 10 pro [64bit]
| Software Version | OTD version v0.4.1
| Tablet           | Tablet gaomon m10k pro

## Diagnostic Information

(
[gaomon m10k pro.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6948612/gaomon.m10k.pro.txt)
)

Mrcubix, 2021-08-07:
[M10K Pro.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6949246/M10K.Pro.zip)
extract the content of this archive in `[OTD Folder]\Configurations\Gaomon` (if you use portable as you should)
else you'll find the configuration folder at this path : `%localappdata%\opentabletdriver\bin\Configurations\Gaomon` (if you used the installer)

no6hope, 2021-08-09:
thanks, now it is working :D

closed 2021-08-09 by X9VoiD