OpenTabletDriver/OpenTabletDriver #1304: Huion tablet not being detected

AlphaWoofle, 2021-07-31:
## OpenTabletDriver cant detect my Huion HS64 Graphics Tablet 

Hello, I've been trying to use OpenTabletDriver on my Huion HS64 Graphics Tablet but its not being detected at all. Im on Linux Mint and i cant get it to work with my Huion tablet
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Linux Mint
| Software Version | 20 "Ulyana" - Cinnamon
| Tablet           | Huion HS64

gonX, 2021-07-31:
Please update the Software Version in System information to say OTD's version, rather than your OS' version

Mrcubix, 2021-07-31:
Also Please provide a diagnostic file, You seem to have deleted the most important part of this issue for some reason.
Here is how to export a diagnostic file, and add it to your first message:

- Download the right build according to your OS from the [Latest Release](https://github.com/OpenTabletDriver/OpenTabletDriver/releases/tag/v0.5.3.3).
- Open the UX using `opentabletdriver` or open it using the desktop icon (if one exist)
- Read the guide if it's your first time trying this driver, and if after closing the guide, your tablet isn't detected, 
click on `Help` -> `Export Diagnostic` -> Save it wherever you want -> add .txt add the end
- Add the following template at the end of your first message:
```
## Diagnostic Information
<!-- Export diagnostics and attach or paste below -->
<!-- In the GUI, click Help -> Export Diagnostics -->
```
- Then upload your diagnostic file by either dragging and dropping it, selecting or pasting them.

gonX, 2021-07-31:
@Mrcubix bug report templates do not currently have diagnostics information as a part of the template - support requests do. But yes, a diagnostics file would be nice here

zooxs, 2021-09-22:
> Also Please provide a diagnostic file, You seem to have deleted the most important part of this issue for some reason.
> Here is how to export a diagnostic file, and add it to your first message:
> 
>     * Download the right build according to your OS from the [Latest Release](https://github.com/OpenTabletDriver/OpenTabletDriver/releases/tag/v0.5.3.3).
> 
>     * Open the UX using `opentabletdriver` or open it using the desktop icon (if one exist)
> 
>     * Read the guide if it's your first time trying this driver, and if after closing the guide, your tablet isn't detected,
>       click on `Help` -> `Export Diagnostic` -> Save it wherever you want -> add .txt add the end of the file name
> 
>     * Add the following template part at the end of your first message:
> 
> 
> ```
> ## Diagnostic Information
> 
> 
> ```
> 
>     * Then upload your diagnostic file by either dragging and dropping it, selecting or pasting them.

i'v same issue, which one diagnostic file should you upload?

Mrcubix, 2021-09-22:
in your own issue: 
[image]

InfinityGhost, 2021-10-30:
Stale support request, assuming it has been resolved. This can be reopened in the case this was a mistake.

closed 2021-10-30 by InfinityGhost