OpenTabletDriver/OpenTabletDriver #1314: Support Request: Artisul A1201

azaz08967565, 2021-08-04:
# Support Request
I bought Artisul A1201, and this tablet is not supported, please add this tablet configuration into the tablet driver. This tablet have 440 pps rate.

SnowLire, 2021-08-04:
In OTD, go to Help > Export Diagnostic, zip the file and send it here.

azaz08967565, 2021-08-04:
[Artisul A1201.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6931485/Artisul.A1201.zip)

Mrcubix, 2021-08-15:
[Artisul A1201.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6988924/Artisul.A1201.zip)
- Create a new folder in `[OTD Folder]\Configurations` called Artisul and extract the content of this zip file in it.
- Open OTD's UX, go to Tablets > Tablet debugger
- Move your pen around, you'll see that some values appear, on the right you'll see the value for position.
- I want you to tell me the maximum value for position you can get **by moving to the bottom right corner**

azaz08967565, 2021-08-16:
[51689,34308]

Mrcubix, 2021-08-16:
is this the complete max you can get?

azaz08967565, 2021-08-16:
Yes, im pretty sure

Mrcubix, 2021-08-16:
[Artisul A1201.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6989463/Artisul.A1201.zip)
- Remove the old config, extract this one to Configurations\Artisul
- Go to Tablet > Device String Reader > Dump all > save the file somewhere > add .txt at the end if necessary when uploading it here

azaz08967565, 2021-08-16:
[Artisul A1201.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6989493/Artisul.A1201.txt)

Mrcubix, 2021-08-16:
[Artisul A1201.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6989507/Artisul.A1201.zip)
same thing, except everything should be working now

azaz08967565, 2021-08-16:
In this file I have 2 configurations, is this ok? Or i need to use the new one only?

azaz08967565, 2021-08-16:
[Artisul A1201.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6989519/Artisul.A1201.txt)

Mrcubix, 2021-08-16:
as i said, replace the old one, and test if your tablet work

azaz08967565, 2021-08-16:
Sorry, my bad, its works.

Mrcubix, 2021-08-16:
I need to you test everything from position, buttons, everything

azaz08967565, 2021-08-16:
all buttons are working, cursor is moving normaly, i think its shoud be fine

InfinityGhost, 2021-10-30:
Stale support request, assuming it has been resolved. This can be reopened in the case this was a mistake.

closed 2021-10-30 by InfinityGhost