OpenTabletDriver/OpenTabletDriver #1315: Minimize to tray on startup

EnanoFurtivo, 2021-08-04:
This would be a really nice addition. Personally I place this program in the startup folder because I always end up using it. Having to minimize it twice a day gets old fast.

[image]

InfinityGhost, 2021-08-04:
Not all platforms use a tray icon (Linux is an example) so it's not really something I want to implement. However, we do provide an [option](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/General-FAQ#ui) to start the program minimized

closed 2021-08-04 by InfinityGhost

Mrcubix, 2021-08-04:
You can also go into the shortcut properties and in `Run`, select minimized

EnanoFurtivo, 2021-08-04:
Thanks for the suggestion, I ended up using -m in the shortcut as InfinityGhost stated.