OpenTabletDriver/OpenTabletDriver #1319: Gaomon M10K not detected

WaifuMat3rial, 2021-08-05:
# Support Request

[image]
[image]

## What I've already tried

Mrcubix, 2021-08-05:
Please follow the following template

```template
## Description
<!-- Describe the issue below -->

## System Information:
<!-- Please fill out this information -->
| Name             | Value |
| ---------------- | ----- |
| Operating System | OS here
| Software Version | OTD version here
| Tablet           | Tablet here

## Diagnostic Information
<!-- Download OTD in the Release section -->
<!-- Start the UX -->
<!-- Click on Help -> Export diagnostics and attach (add `.txt` if necessary) or paste below -->
```

gonX, 2021-12-06:
Closing as stale support request. Your tablet is supported.

closed 2021-12-06 by gonX