OpenTabletDriver/OpenTabletDriver #1322: Support Request:

ItzKrypticOsu, 2021-08-07:
# Support Request
 g s620 not being detected...
 i downloaded otd, deleted the hawku and gaomon drivers. 

## What I've already tried

i tried to open a config but it says [Detect:Error]	The configuration directory 'C:\Users\corn_\Desktop\Configurations\Configurations' does not exist.
ive tried to export diagnostic but it does not work, just opens blank document

## Diagnostic Information

[image]

Mrcubix, 2021-08-07:
Do not open multiple issues, extract the zip file content or make sure the configuration folder and daemon are in the same directory

Mrcubix, 2021-08-07:
also a diagnostic file is necessary for further support if your tablet isn't working after fixing your previous issue:

`Help > Export Diagnostics`

ItzKrypticOsu, 2021-08-07:
how do i make sure the config fold and daemon are in the same directory?

ItzKrypticOsu, 2021-08-07:
ive tried export diagnostics and i open the notepad where it is supposed to show the diagnostic and its just blank

ItzKrypticOsu, 2021-08-07:
even tried the winusb cleanup file in the video you made and it made no difference

Mrcubix, 2021-08-07:
- Download OTD: https://github.com/OpenTabletDriver/OpenTabletDriver/releases/download/v0.5.3.3/OpenTabletDriver.win-x64.zip
- Extract OTD to a separate folder located on your desktop or idk.
- Config folder should be found
- If tablet isn't detected : `Help > Export diagnostics`

ItzKrypticOsu, 2021-08-07:
i finally got it thank you for your time Mrcubix

closed 2021-08-07 by ItzKrypticOsu