OpenTabletDriver/OpenTabletDriver #1325: driver disables my mouse entirely

0jeannedearc, 2021-08-09:
I'm using the most recent opentabletdriver and it seems whenever I plugin my tablet to one of my usb slots (the one I have my mouse put into) I cant use the buttons on the pen, even though I bound them in the buttons tab in the driver, however the main issue is when I unplug my tablet and replug my mouse in my mouse is completely disabled, I can't click or move the cursor around at all, I'm just wondering whats going on here, I'm using a wacom 472 and windows 10

X9VoiD, 2021-08-09:
OTD doesn't have anything to do with mouse, and it doesn't even have any access to the mouse at all since that requires kernel-level access. OTD is a user-mode driver. Do you have Valorant installed?

0jeannedearc, 2021-08-09:
Yeah I do, is it something to do with the anticheat blocking something in otd?

X9VoiD, 2021-08-09:
Yes, Vanguard only allows one input source. Closing this non-issue.

closed 2021-08-09 by X9VoiD