OpenTabletDriver/OpenTabletDriver #1327: Support Request: Can't install the requirement for VMultiMode

ElboyYT, 2021-08-09:
# Support Request:

Installing the requirement for VMultiMode

I was said that i need VMultiMode to use my tablet as a mouse in Valorant.
I was given this information and i should install it with the requirement. But i can't install it and the OTD wont restart when i try to install it. 
Basically: I assume it won't install because the OTD wont Shut down if i try to install it. 
## What I've already tried 
used the "uninstall hiddriver" and tried to install it

## Diagnostic Information
/

Mrcubix, 2021-08-09:
uninstall vmulti from your previous driver (i assume it was hawku drivers) and you should be able to install OTD's vmulti and use the VmultiMode Plugin

ElboyYT, 2021-08-09:
still won't install the Vmulti

Mrcubix, 2021-08-09:
just come to the discord server, it will be 100 times easier to troubleshoot thing there https://discord.gg/P3GYdUS

ElboyYT, 2021-08-09:
alright, joined

Mrcubix, 2021-08-09:
Resolved: Vanguard known issue

closed 2021-08-09 by Mrcubix