OpenTabletDriver/OpenTabletDriver #1329: gaomon s620 not working anymore

Ominous15, 2021-08-10:
## Description

I use OTD all the time for osu but its not juist not wokring in general, i cant move to cursor or anything, i can tap with the pen but thats it
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         |
| OpenTabletDriver Version | 
| Tablet                   |

Mrcubix, 2021-08-10:
Does `Tablets > debugger` show the pen movement?

durbin-164, 2021-08-12:
@Mrcubix  Yes, For may case Debugger working But no course move for actual use.

Mrcubix, 2021-08-12:
- Make sure User Account control is enabled to one of the top options, reboot.
- Make sure you're not using the `Administrator` account,
- Make sure Valorant isn't running in the background (kill it in task manager)

durbin-164, 2021-08-12:
@Mrcubix Thanks a lot. After rebooting my os. It working now. Really greatfull to you.

I am new in tablet and OTD. Would you like to suggest me some useful addons that help me for writing and drawing?

Mrcubix, 2021-08-12:
recommended: Reconstructor .43
preference: smoothing or anti-chatter, i won't tell you how to setup these as they have a wiki and it's preference

closed 2021-08-12 by Mrcubix