OpenTabletDriver/OpenTabletDriver #1331: Support Request: need help setting up specific tablet

26wmsa, 2021-08-11:
# Support Request
im trying to set up the open tablet driver for the Parblo nino s but it keeps crahsing whenever i trty to create a config
i press save and it just crashes and opens but the config doesnt save. the tablet isnt coming up as detected either.

## What I've already tried
deleting, restarting unplugging tablet, everything i can really since theres no videos on this issue :(

doubledge-spec, 2021-08-11:
You need to use correct templates for submitting requests.

doubledge-spec, 2021-08-11:
this is a configuration request, so use the configuration request template so we have info

SnowLire, 2021-08-11:
> # Support Request
> im trying to set up the open tablet driver for the Parblo nino s but it keeps crahsing whenever i trty to create a config
> i press save and it just crashes and opens but the config [...]

You need to download a dev version [from here](https://github.com/OpenTabletDriver/OpenTabletDriver/actions?query=event%3Apush+is%3Acompleted+branch%3Amaster). Get one that's correct for your OS, extract all the files into a folder and run the UX.

26wmsa, 2021-08-11:
i downloaded it but it doesnt open it goes through the guide and it just closes after a second

26wmsa, 2021-08-11:
im litterally so confused on what you meant by the format this is my 1st time using github so i dont understand. i made a new issue bc the guy said i submitted it wrong. Again, i couldn't make a config the OTD crashed when i pressed save every single time

Mrcubix, 2021-08-11:
*just do the other steps that i provided*

26wmsa, 2021-08-11:
you keep using the word config and its getting frustrating becasue i dont know weather you are talking about in the OTD or on github. like i said im new to this so none of it is really making sense

Mrcubix, 2021-08-11:
you use the word config and yet you don't understand? i'm talking about the OTD configuration you made, that's is not made properly and that is crashing OpenTabletDriver as we speak, so here are the steps you need to do again before can troubleshooting anything:

- DON'T open any other issues.
- Delete the Tablet configuration you made (you'll be able to open OTD after that).
- Export a diagnostic (in the Interface: `Help > Export diagnostics`), add .txt at the end.
- Edit your original message in this issue and Fill the template:
```
# Support Request
<!-- What do you need help with? -->
<!-- Be descriptive and use images or video where possible. -->

## What I've already tried
<!-- Reference any wiki pages you've already tried -->

## Diagnostic Information
<!-- Export diagnostics and attach or paste below -->
<!-- In the GUI, click Help -> Export Diagnostics -->
```
- Upload the diagnostic file where it belong in the template

26wmsa, 2021-08-11:
for the third time, i did mention that the config ISNT saving every time it crashes. its when i try to save it thats causing to crash and when i open it again its not there

Mrcubix, 2021-08-11:
*just do the rest of the steps then?*

26wmsa, 2021-08-11:
again....... i said i dont unerstand whats wrong with the format. Ive filled everything out no ?

Mrcubix, 2021-08-11:
The `Diagnostic information` is obviously missing looking at the `support request` template, therefore, we need you to fill that last part, by copying that part of the template, editing your original message, pasting it, exporting a diagnostic file and uploading it

26wmsa, 2021-08-11:
done. i pasted the diagnostic info in as the .json was not supported

Mrcubix, 2021-08-11:
like i told you before, you could had added .txt add the end, but anyway, looking at it rn

Mrcubix, 2021-08-11:
So, download this [Dev Build](https://github.com/OpenTabletDriver/OpenTabletDriver/suites/3448273559/artifacts/81821822)
then extract it in it's own folder, start the interface and it should work as far as we are aware

26wmsa, 2021-08-11:
i did that exact thing and this version doesnt even open

Mrcubix, 2021-08-11:
delete `%localappdata%/opentabletdriver`

26wmsa, 2021-08-11:
ok i did it and the tablet now being recognised :) . how can i tell that the tablet is propperly working and connected because i saw that the Parblo Nino S recently got added to the support list ?

Mrcubix, 2021-08-11:
move around, hover, drag, test the buttons binds, that kind of things

26wmsa, 2021-08-11:
ok i did and i think it works 👍 many thanks. so is this giving me the least input delay ?

X9VoiD, 2021-08-11:
Yes, unfortunately the tablet itself have hardware smoothing and OTD have no control over that. This smoothing is baked directly in the hardware itself as the name suggests, introducing high input latency. Will close since issue is resolved now.

closed 2021-08-11 by X9VoiD