OpenTabletDriver/OpenTabletDriver #1332: Support Request:

26wmsa, 2021-08-11:
# Support Request
Need help setting up the Parblo Nino S in the Open tablet driver.
Every tiem i try to make a new config it just crashes when i press save and when i re-open, its not been saved.

## What I've already tried
I've tried to re-install OTD and re-installed the .NET runtime. Ive tried installing a different way all have had no luck

## Diagnostic Information
there was none

doubledge-spec, 2021-08-11:
you need to download a dev version. from here: https://github.com/OpenTabletDriver/OpenTabletDriver/actions?query=event%3Apush+is%3Acompleted

26wmsa, 2021-08-11:
what am i downloading

doubledge-spec, 2021-08-11:
choose the latest result, and grab the windows build from there.