OpenTabletDriver/OpenTabletDriver #1337: Support Request: VEIKK S640 not being being detected but driver not working

Jasofish, 2021-08-13:
# Support Request
My Veikk tablet couldn't be detected by the driver in the first place, my friend who uses it gave me a json (Veikk S640 V2)  file that made the driver detect my tablet. However, even if it says "Found Tablet", the driver still says no tablet found in the output tab.
[image]
[image]

## What I've already tried
I've tried the Json v2 file, I also tried deleting the standard drivers and I tried unplugging and plugging my tablet back. (I'm a new tablet user so I don't know much of what I can do)

## Diagnostic Information
[support request.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6979518/support.request.zip)

Mrcubix, 2021-08-13:
This tablet is currently only supported in `master`, plugins will have to be downloaded from their respective repositories and may not work (eg: windows ink, vmultimode, ...)

First, delete `%localappdata%/opentabletdriver`,

Then Download this [Dev Build](https://github.com/OpenTabletDriver/OpenTabletDriver/suites/3448273559/artifacts/81821822)
Extract the zip file to it's own folder, start `OpenTabletDriver.UX.Wpf.exe`, your tablet should then be detected and should be working as far as we are aware.

gonX, 2021-12-06:
Closing stale support request. S640 works on `master` as mentioned.

closed 2021-12-06 by gonX