OpenTabletDriver/OpenTabletDriver #1338: Support Request: KAMVAS 22 PLUS

Diediediediediediedie, 2021-08-13:
# Support Request

## What I've already tried

## Diagnostic Information

[image]
[image]

SnowLire, 2021-08-13:
Fill out the support template properly. There are steps written for you to follow.
And don't open the same issue twice.

Diediediediediediedie, 2021-08-13:
Sorry~ I don’t know what a template is, but I know it doesn’t support KAMVAS 22 PLUS, so I want to ask for support.

SnowLire, 2021-08-13:
Then it'd be easier if you join the discord server. You can leave this issue open until it's resolved.

Diediediediediediedie, 2021-08-13:
Yes~ I joined discord but I don’t know where to ask

SnowLire, 2021-08-13:
In #help-windows.

PR #1340 (2021-11-04): Add Kamvas 22 plus Configuration
##Verification:

https://discord.com/channels/615607687467761684/615611007951306863/875836307144851468

## Further support provided to user:

https://discord.com/channels/615607687467761684/789348845372178482/875840873471033384

Fix #1338
PR #1679 (2021-12-13): Update configurations and parsers
- Closes #1101
- Closes #1282
- Closes #1338
- Closes #1401
- Closes #1435
- Closes #1448
- Closes #1500
- Closes #1545
closed 2021-12-13 by InfinityGhost