OpenTabletDriver/OpenTabletDriver #1339: filter window does not appear

Skygger1, 2021-08-13:
## Description

filter window does not appear, I have already tried to uninstall and reinstall enable uac but it does not appear (Google traduction)
## System Information:
Intal core I5 ram 4gb 64 bits

| Name             | Value |
| ---------------- | ----- |
| Operating System   widows 10     |
| OpenTabletDriver Version |
| Tablet XP-pen star 03 v2                 |

Mrcubix, 2021-08-13:
If you mean the Filter tab,  it's just a separate tab, not a window, and shouldn't be an issue.
if you mean Plugin > Plugin manager, make sure of the following:

- Enable user account control to one of the top 2 option 
- Don't use the `Administrator` account
- Don't use a debloated OS

Skygger1, 2021-08-14:
i switch to an user account and works thanks !

closed 2021-08-15 by Mrcubix