OpenTabletDriver/OpenTabletDriver #1342: Wacom CTL 472 N - no tablet detected

tinutt, 2021-08-14:
## Description

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Windows 10         |
| v0.5.3.3 |
| wacom ctl 472 N                 |

SnowLire, 2021-08-14:
Please provide a diagnostic file in OTD by doing `Help > Export diagnostic` and send it zipped.

tinutt, 2021-08-14:
how do i send it zipped? sorry im kinda dumb

SnowLire, 2021-08-14:
Right click the file > Send to > Compressed folder.
You can also just change the file extension from .json to .txt and send it as that.

tinutt, 2021-08-14:
[issue.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6986301/issue.txt)

SnowLire, 2021-08-14:
You have to extract the UX, Daemon and Configuration folder into its own folder, preferably on your Desktop and ***NOT*** run them straight from the zip file.

tinutt, 2021-08-14:
it does the same thing
if you need a new diagnostic from where i extracted it-
[3.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6986322/3.txt)

SnowLire, 2021-08-14:
Kill every OTD related process in Task Manager and run the extracted UX again.

tinutt, 2021-08-14:
its fixed, TYSM <3

closed 2021-08-14 by tinutt