OpenTabletDriver/OpenTabletDriver #1345: Bindings don't work (Auxiliary Buttons) XP Pen deco mini7

bespokecomp, 2021-08-15:
## Description

Trying to create bindings with the 8 Auxiliary Buttons.  Doesn't do anything.  The buttons on the stylus is ok.
I tried to set Auxiliary Button 1 as a binding to Delete.  Can't do it.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         |Windows 10
| OpenTabletDriver Version |0.5.3.3
| Tablet                   |XP Pen deco mini7

bespokecomp, 2021-08-15:
[XPPen_deco_mini7_diagnostic2.json.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6988298/XPPen_deco_mini7_diagnostic2.json.zip)

Mrcubix, 2021-08-15:
are you sure you're pressing the right aux key? aux 1 could not be the first aux key on the left or right bu example.

bespokecomp, 2021-08-16:
> 
> 
> are you sure you're pressing the right aux key? aux 1 could not be the first aux key on the left or right by example.
> also does any other key work.

After reading your suggestion, i tried all 8 keys binded to be keys A -H.
I opened notepad on windows and pressed each key.  I didn't get any output at all.

Not sure if there's anything else I'm missing?

gonX, 2021-12-01:
0.5.3.3 has TabletReportParser registered for the deco mini7 which does not have Aux key support. It's fixed on `master`, so Mrcubix' backports plugin should handle it for you.

Or you can replace `Tablet.TabletReportParser` with `Vendors.XP_Pen.XP_PenReportParser` in the configuration file yourself.

Or use a prerelease version.

gonX, 2021-12-06:
Closing stale support request.

bespokecomp, 2021-12-12:
> 0.5.3.3 has TabletReportParser registered for the deco mini7 which does not have Aux key support. [It's fixed on `master`](https://github.com/OpenTabletDriver/OpenTabletDriver/blob/master/OpenTabletDriver.Configurations/Configurations/XP-Pen/Deco%20Mini7.json), so Mrcubix' backports plugin should handle it for you.
> 
> 
> 
> Or you can replace all instances of `Tablet.TabletReportParser` with `Vendors.XP_Pen.XP_PenReportParser` in the configuration file yourself.
> 
> 
> 
> Or use a prerelease version.

Thanks for that info. 

I'll see if I can sus this out. 
This one here right? https://github.com/Mrcubix/OTD.Backport

And this is a plugin that I add through the normal build and not the dev build, right?

gonX, 2021-12-12:
Yes, the Backport plugin is intended to use with the normal 0.5.3.x build. It should be available in the plugin manager.

Once installed it should be as simple as enabling the plugin and clicking Detect Tablets in OTD (Ctrl+D)

bespokecomp, 2021-12-12:
> Yes, the Backport plugin is intended to use with the normal 0.5.3.x build. It should be available in the plugin manager.
> 
> 
> 
> Once installed it should be as simple as enabling the plugin and clicking Detect Tablets in OTD (Ctrl+D)

Ok. Cheers. I'll give it a go.

jamesbt365, 2021-12-31:
Do you have an update on this?

jamesbt365, 2021-12-31:
Resolved in master.

closed 2021-12-31 by jamesbt365

bespokecomp, 2022-01-04:
> Do you have an update on this?

So far I couldn't get it to work. 
Having loaded a Dev build previously, I had that on my desktop (and didn't want to lose that).

So I tried to run the original OTD from the C drive. 

As far as I know, I was. I installed backports plugin and tried to assign some bindings. Wouldn't work and I believe it then started to conflict with the dev build.