OpenTabletDriver/OpenTabletDriver #1346: Support Request:

mochi39, 2021-08-15:
# Support Request

Hello, Im new to this tablet driver and theres something wrong about it when i try launching it.

This picture below shows when i try starting it, it just exit the app and reopen it (i tried this many times but it still the same) but it doesnt show the inside of it like the picture B that ive shown. I need help please thanks in advance!

[image] (picture A)

[image] (picture B)

mochi39, 2021-08-15:
i already open the daemon, and it just showing me the cmd. (just wanted to show this so that nothing wrong happens on the daemon)

[image]

jamesbt365, 2021-08-15:
what tablet are you using?

Mrcubix, 2021-08-15:
Don't use the installer, use the portable release https://github.com/OpenTabletDriver/OpenTabletDriver/releases/tag/v0.5.3.3 don't forget to extract the zip file to it's own folder

mochi39, 2021-08-16:
> what tablet are you using?

im using the gaomon s620.

mochi39, 2021-08-16:
> Don't use the installer, use the portable release https://github.com/OpenTabletDriver/OpenTabletDriver/releases/tag/v0.5.3.3 don't forget to extract the zip file to it's own folder.
> 
> if it still close when opening the UX then delete `%localappdata%\opentabletdriver` and try again.
> if it persist, let us know in this issue by just posting another message and i'll re-open.

okay, will try it

mochi39, 2021-08-16:
Hello, i already try the portable one but i think its still have the same problem. ( i dont know if i did it right or wrong)

i try to run the daemon and then starting the UX, but it doesnt open/run the app. Then i try to delete the folder on the localappdata and try it again but still the same. (it only shows the Plugins folder on the localappdata)
[image] (picture A)

This picture below shows that the UX app doesnt appear on the task manager when i try running it.
[image] (picture B)

Mrcubix, 2021-08-16:
do `dotnet --list-runtimes` in cmd and paste the result here

mochi39, 2021-08-16:
> do `dotnet --list-runtimes` in cmd and paste the result here

C:\WINDOWS\system32>dotnet --list-runtimes
Microsoft.NETCore.App 5.0.9 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
Microsoft.WindowsDesktop.App 5.0.9 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]

jamesbt365, 2021-12-31:
This is either, a stuck daemon, can be solved by killing it in task manager or restarting your pc, missing dotnet, which we can see that you have above or probably something missing between wpf applications which can be caused by debloating or otherwise damaging your windows install.

closed 2021-12-31 by jamesbt365