OpenTabletDriver/OpenTabletDriver #1347: Support Request: RelativeDragMove

dbot101, 2021-08-15:
Im currently trying to play FPS games with mouse mode enabled but I find that even when hovering my cursor follows my pen. I do not want that. I only want my cursor to move when I press down on the pen tip and drag. I cannot find anything on reducing lift off distance or disabling hover movement 

I found this old hawku drivers post that seems to highlight exactly what I want but I cannot find how to enable this type of feature but it seems even they couldnt solve their problem 
https://github.com/hawku/TabletDriver/issues/670

If anyone has some advice or input they could give me that would be greatly appreciated

Mrcubix, 2021-08-15:
hover distance limiter plugin only work in dev build, Though not sure it's working with the recent changes,
dev build: https://github.com/OpenTabletDriver/OpenTabletDriver/suites/3448273559/artifacts/81821822 (only accessible if logged into github, else 404)
hover distance limiter plugin : https://cdn.discordapp.com/attachments/628651971267788800/876573176480542740/win-x64.7z
(drag and drop this in plugin manager)

dbot101, 2021-08-15:
> hover distance limiter plugin only work in dev build, Though not sure it's working with the recent changes,
> working build: https://cdn.discordapp.com/attachments/628651971267788800/876573176480542740/win-x64.7z

I dragged the "0.1.0_release_Hover_Distance_Limiter" file into my own drivers plugins but I still dont see an option under Bindings, filters, or tools 

How do I get the dev build to work?

Mrcubix, 2021-08-15:
i edited my message previously, just download the build i sent, extract it to its own folder, start the UX, go in the filter tab and set your setting for the Hover distance limiter there

dbot101, 2021-08-15:
> i edited my message previously, just download the build i sent, extract it to its own folder, start the UX, go in the filter tab and set your setting for the Hover distance limiter there

When I launch the UX it opens for 3 seconds shows "Tablet not detected" at the top of the window  then closes itself. It does not do that for the original version i have

closed 2021-08-15 by dbot101