OpenTabletDriver/OpenTabletDriver #1351: Tablet debugger freezes on daemon restart

C0D3-M4513R, 2021-08-16:
## Description

When I restart the driver daemon (but not the UI), the tablet debugger freezes, and needs to be restarted.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 10
| OpenTabletDriver Version | 1e366358d7b8ac24c47c4068abbe5c924f7fdc43
| Tablet                   | MTE-450

TODO: is this also the case, on release versions?

SnowLire, 2021-08-16:
I'm certain that happens because the debugger don't actively look for another daemon to connect to, unlike the UX. This is such a specific scenario that I doubt there's the need to fix.

C0D3-M4513R, 2021-08-16:
If the UX knows the active Daemon, why can't the tablet debugger use a reference, to that variable, the UX keeps?
There should be no need, for multiple Daemon connections, or am I mistaken?

SnowLire, 2021-08-16:
The UX can probably be used as a reference for the debugger's connection to the daemon, although this is caused by such a specific chain of actions that it's very low priority for us. You are welcomed to submit a PR if you feel like fixing it.

PR #1352 (2021-08-30): Fix tablet debugger not completely reconnecting
- Fixes #1351.
closed 2021-08-30 by InfinityGhost