OpenTabletDriver/OpenTabletDriver #1353: CTH-460 non operable on Arch Linux

parkerlreed, 2021-08-17:
## Description
Tablet is detected but no inputs present in monitor or evdev
Kernel appears to attach a dunmy mouse to touch input via hid-generic which can't be blocked.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         |Arch Linux kernel 5.13.10
| OpenTabletDriver Version |v0.5.3.1.r399.g1e366358-1
| Tablet                   |CTH-460

https://gist.github.com/parkerlreed/14f0b25de31f505312251aaa4b96e0aa

```
[  889.651658] usb 1-10: new full-speed USB device number 9 using xhci_hcd
[  889.782632] usb 1-10: New USB device found, idVendor=056a, idProduct=00d1, bcdDevice= 1.06
[  889.782644] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  889.782650] usb 1-10: Product: CTH-460
[  889.782654] usb 1-10: Manufacturer: Wacom Co.,Ltd.
[  889.791134] input: Wacom Co.,Ltd. CTH-460 Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:056A:00D1.0009/input/input29
[  889.791946] hid-generic 0003:056A:00D1.0009: input,hiddev97,hidraw2: USB HID v1.00 Mouse [Wacom Co.,Ltd. CTH-460] on usb-0000:00:14.0-10/input0
[  889.794987] hid-generic 0003:056A:00D1.000A: hiddev98,hidraw3: USB HID v1.00 Device [Wacom Co.,Ltd. CTH-460] on usb-0000:00:14.0-10/input1
```

gonX, 2021-12-06:
Is this still the case on latest master? A lot has changed since then.

If it's the case please export diagnostics from OpenTabletDriver (Help -> Export Diagnostics in GUI) and upload them here.

parkerlreed, 2021-12-06:
Thanks for the reminder. I should be able to test later if I can find the pen. Will let you know.

gonX, 2021-12-20:
Here's to hoping you found the pen :)
A few common mistakes most people seem to do on Arch: not having the daemon started (`systemctl --user start opentabletdriver`) and not replugging the tablet (or running `udevadm trigger`) since installing OTD.

jamesbt365, 2022-02-28:
If the previous thing didn't solve it then [this](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8153871/CTH-460.json.txt) config should do so, download it, remove the `.txt` at the end leaving just `CTH-460.json` and then move the file to `/usr/share/OpenTabletDriver/Configurations` This config will only work from 0.6.0.x onwards.

closed 2022-02-28 by jamesbt365