OpenTabletDriver/OpenTabletDriver #1355: Virustotal flags malware in windows portable v0.5.3.2 and above

pushkine, 2021-08-17:
Related #883 
* **v0.5.3.1** Safe https://www.virustotal.com/gui/file/b738d7f044d0be05013c64c0d3c490a7e2c803a897875df63147cdc0b1318acd/detection
* **v0.5.3.2** Unsafe https://www.virustotal.com/gui/file/99c2c9754cd9f637a890cae3ed55ab0e7b7cded7b9ed8ba8837365e2092e6d40/detection
* **v0.5.3.3** Unsafe https://www.virustotal.com/gui/file/42647f688074b8b1e15345ed08c8a9f2e682981a44c12b1ac5e6037876c911a3/detection

InfinityGhost, 2021-08-17:
These are all false positives, we basically have to submit a request to have a manual review performed in order to get them to go away. It's inconvenient and plagues open source software as a whole.

X9VoiD, 2021-08-18:
One or two detections doesn't mean the program is unsafe. That just means the AV is shit and reports false positive. Open this issue again when you see atleast 40% of the AVs reporting OTD as malware.

closed 2021-08-18 by X9VoiD