OpenTabletDriver/OpenTabletDriver #1357: Auxiliary and pen button note fields

caseyjp11, 2021-08-18:
## Description
As the subject indicates, it would be nice to have an additional field to save notes on the button assignments in the auxiliary tab and pen tab as well as a field to indicate the actual Wacom ID. For many of us coming over from Wacom or digimend with Linux, we have had to use scripts to set up this stuff and while the simplicity of OTD's binding numbers is nice, it can be royally confusing at times when the actual ID doesn't match up to the information that xsetwacom requires. (see the mock-up for details)

As well, having the ability to save the assignments with a 'save as' to further customize settings for different applications would definitely be a help. (I believe this can be done already, but it is not intuitive.)

1st shot is the mock-up, 2nd shot is a snippet from my script file used with the digimend stuff:

[image]

[image]

InfinityGhost, 2021-08-18:
After some discussion in the OpenTabletDriver Discord Guild, we've decided we probably won't add notes however we will add a configuration specification to manually specify ordering of buttons in the GUI.

caseyjp11, 2021-08-18:
Well every little bit helps. The 'save as' would be more important to me unless that is elsewhere. Once I've got a set of keys 'bound' for an app, I normally don't look at it again other than to run the script for the app (xsetwacom stuff), so I just create scripts for the apps I'm using. Having the ability to switch on the fly via the GUI would be outstanding.  
--AND, if I'm completely braindead and that capability is already in the app, please let me know where.
Again, thank you for the awesome work that is being done here! :-)