OpenTabletDriver/OpenTabletDriver #1358: Bindings per application

d3xMachina, 2021-08-18:
I would like to have the possibility to bind the buttons of the tablet differently depending on the current active application. This would be really useful.

So in the "Bindings" tab, you would first select an application (or "All"/"Default") and then you would change the corresponding bindings.

This is a feature that already exist in the drivers of Wacom (unfortunaly I don't use a wacom tablet) : 
[image]

Thanks.

InfinityGhost, 2021-08-18:
This is extraordinarily difficult to pull off consistently between all supported platforms so this really will sit on the backlog for a while

gonX, 2021-08-19:
As a temporary workaround, you can export your settings and load them manually as needed for now.

On Linux, I use this launch script to launch Krita - it's not per-app detection, but it principally loads my artist profile while starting Krita, and then loads my non-artist profile when Krita is closed:
```bash
#!/usr/bin/env bash

ARTIST_PROFILE=/home/gonx/documents/tablet-artist.json
ABSOLUTE_PROFILE=/home/gonx/documents/tablet-osu.json

notification(){
    notify-send -a 'Krita Launcher' -i /usr/share/pixmaps/otd.png "$@"
}

otd load "$ARTIST_PROFILE" &

krita &

wait %1
notification "Artist Mode enabled"

wait %2
otd load "$ABSOLUTE_PROFILE"
notification "Artist Mode disabled (osu profile loaded)"
```

On Windows you can likely achieve the same with a batch file by doing something like:
```bat
\path\to\otd.console.exe load \path\to\artist-profile.json
\path\to\krita.exe
\path\to\otd.console.exe load \path\to\non-artist-profile.json
```

d3xMachina, 2021-08-19:
Thanks for the help!

I'm on Windows and the "OpenTabletDriver.Console.exe" is not included in the installation. So I compiled it myself but when I run it (as admin or not) I get the error "Unhandled exception. System.UnauthorizedAccessException: Could not connect to the pipe because it was not owned by the current user."

I wanted to use your idea to make a autohotkey script that load a profile when the active window change as a workaround with a shell hook (call to RegisterShellHookWindow then RegisterWindowMessage with "SHELLHOOK" as a parameter)

X9VoiD, 2021-08-19:
Make sure that you're not running the Daemon as admin. As the error states, the user should match, so either you run both as admin, or you run both as user and in this case you should run both as user.

d3xMachina, 2021-08-19:
Thanks it works now. For some reason my console was run as admin when I tought it was not.

InfinityGhost, 2021-08-28:
> Thanks it works now. For some reason my console was run as admin when I tought it was not. Btw, is there a reason why the console is not included in the release for the windows platform ?

Mostly for filesize, it's rarely used even by the Linux users so we just don't include it on Windows. If it becomes something people actually want in the release we can include it (maybe separately)

d3xMachina, 2021-08-30:
I have implemented the bindings per application on [my fork](https://github.com/d3xMachina/OpenTabletDriver/commit/c583734eee23e2d61d9886842aafe7178e1eb041) for windows only . Since I'm not familiar with c# it took me a bit of time. I hope it will help for a future implementation.

I used a shell hook with a hidden form to hook the window messages for the activation of windows.
To use it, you have to put your profiles in the "profiles" folder inside the OpenTabletDriver path (or create it if it doesn't exist). Then you rename your profiles to the application name you want to bind while keeping the ".json" extension. (example : notepad.json for notepad.exe).

I'm not familiar with the linux API and GTK so I can't help with that though. I also didn't implement the UI or commands to retrieve the list of profiles in the daemon.

PS : About the console application, I didn't even know about it until it was mentioned by gonX. I think it's not in the FAQ and may be a reason why people don't ask for it.

closed 2025-12-06 by gonX