OpenTabletDriver/OpenTabletDriver #1366: Stuck on full area

uhmmoyo, 2021-08-21:
## Description
after a windows update - launching open tablet driver seems to do nothing. It detects the tablet fine, but it's stuck on full area and pen reacts to windows ink default settings.

i'll attach a diagnostic report 

## System Information:
| Operating System | Windows
| OpenTabletDriver Version | v0.5.3.3
| Tablet | Wacom CTH-480

uhmmoyo, 2021-08-21:
[8.21.21Diagnostic.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7026364/8.21.21Diagnostic.txt)

Mrcubix, 2021-08-22:
uninstall wacom drivers

uhmmoyo, 2021-08-23:
> uninstall wacom drivers

I don't have the wacom drivers installed. I re-installed them and uninstalled them again successfully and still nothin.

Mrcubix, 2021-08-23:
export diagnostic again

X9VoiD, 2021-09-26:
Will close stale issue.

closed 2021-09-26 by X9VoiD