OpenTabletDriver/OpenTabletDriver #1367: Scrollwheel support

C0D3-M4513R, 2021-08-22:
## Description
The MTE-450 has a scroll wheel. I would like, for some kind of implementation, that lets the scroll wheel be bound, to e.g. Mouse wheel? or windows wheel.

caseyjp11, 2021-09-08:
Adding in my request for the HS610's wheel.

gelatinbomb, 2023-01-18:
I'm curious about something. Can't we have the wheel function as two auxiliary buttons so you can choose what key is gonna be pressed when the wheel turns up or down?

I did a test using evtest and the wheel is detected as buttons so If I rotate the wheel up or down evtest detects it as a quick push and release of the chosen key. Not only that, I tested all the XP Pen devices on my system and the one that holds the wheel is in fact the XP Pen Virtual Keyboard.

I don't know how the other tablets would handle the behavior of the wheel but maybe as a temporal or easy solution the wheel coould be understanded on the Opentabletdriver as a set of auxiliary buttons you can bind.

jamesbt365, 2023-01-18:
Having the wheel function as 2 auxiliary buttons is not proper support but could be done. Proper support should be within the next version for some tablets.

andreluizrcn, 2023-05-30:
Hello! hope this message reaches you well, I would like to know if the Veikk A15 pro have some kind of a future wheel support !!! thanks :)

iamalsonothere, 2024-01-05:
Adding in my request for the XP-PEN Deco 03's wheel.

jamesbt365, 2024-04-27:
FYI asking for specific tablets wheels to be supported when you haven't provided data for how that tablets wheel functions is futile because there isn't actually anything to go off for supporting said wheel.

In the future unless you are going to provide a data recordings for all the wheels functions that clearly describe what is going on, not just randomly dumping a data recording then ghosting us, please don't ask for specific tablets wheels, they will all eventually be supported when OTD supports wheels.

iamalsonothere, 2024-04-27:
Sorry, I was under the impression that it was somewhat standardized

Heres a recording of my tablets wheel being turned one way for half the file and the opposite way in the other half

[tablet-data.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/15139177/tablet-data.txt)

Nukley, 2024-05-01:
Attached is a file containing a relevant snip of data, along with a breakdown on what it means. 

Attached below that is a gif of me showing a really bad demo program on how these values work.

[artist13.3pro-tabletbreakdown.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/15173832/artist13.3pro-tabletbreakdown.txt)
[image]

jamesbt365, 2024-06-01:
Considering this isn't closed, the PR that partially implements this isn't merged, and the fact that you still can't configure it. No there isn't an update on this.

No wheel support exists on **any** tablets in this driver. You can see this on our website where all tablets that have wheels are marked as missing wheel support. 0.7 is not close to being completed at this time and you can see the milestone [here](https://github.com/OpenTabletDriver/OpenTabletDriver/milestone/19).

Asking if there is any updates on this when everything points to there not being will not speed support for this feature up, if you need it and there isn't an alternative that fits your needs then use the PR previously mentioned or create your own implementation and open a pull request for it.

We have limited resources at our disposal and this is the fate of most FOSS, if you want it done now write the code for it and open a pull request.

This goes for everyone finding this issue in the future, please do not comment asking for updates, it does not speed things up.

Please only comment if you have data recordings that clearly shows how data is reported from the wheels on your tablet or something else that could be deemed useful.

StandingPadAnimations, 2024-07-12:
Hey y'all, I got a few recordings from the XP-Pen Deco 03 tablet (Arch Linux with Artist Mode set as the output), which just came in the mail earlier today. I've taken 2 recordings:
- Turning the wheel clockwise: [clockwise-deco-3-tablet-data.txt](https://github.com/user-attachments/files/16189082/clockwise-deco-3-tablet-data.txt)
- Turning the wheel counterclockwise: [counterclockwise-deco-3-tablet-data.txt](https://github.com/user-attachments/files/16189083/counterclockwise-deco-3-tablet-data.txt)

Based on the output, it seems like when turning the wheel clockwise, these sequence of bytes occur (opposite order for counterclockwise):
```
02 F0 00 00 00 00 00 00
02 F0 00 00 00 00 00 40
02 F0 00 00 00 00 00 C0
02 F0 00 00 00 00 00 80
```

Mint-me123, 2024-09-20:
Before I found this great opentabletdriver setup I used xsetwacom and I managed to set up the PTH-660 Wacom up on Linus Mint 21.3 with all the 8 buttons and the scroll wheel with the 4 mode center button. I am a null (as the french say) so I combined 2-3 gists from Github (not pretty) and it allowed me to set the scroll ring/wheel for instance to zoom in and out in steps, change brush size...It's more like having extra buttons than a true scrolling effect however it is intuitive to zoom in and out using the ring/wheel. On one PC I have left it with xsetwacom and the other I installed the opentabletdriver (which is so easy to use). On the PC xsetwacom cant see the PTH-660 anymore. My Q: I don't suppose there is a way to use both at the same time?? The easy GUI of opentabletdriver and the xsetwacom ring/wheel setting. Probably not but I thought Id ask

Everspace, 2024-12-04:
I believe that some wheels are actually just funny buttons. My XPPen Artist 24 Pro only produces the following four events types whenever a discrete step in the wheel happens (audible/tactile click). 

```
{ 02 F0 00 00 00 00 00 01 00 02 00 00 } Left Wheel Clockwise
{ 02 F0 00 00 00 00 00 02 00 02 00 00 } Left Wheel Counterclockwise
{ 02 F0 00 00 00 00 00 10 00 02 00 00 } Right Wheel Clockwise
{ 02 F0 00 00 00 00 00 20 00 02 00 00 } Right Wheel Counterclockwise
```

Lucretia, 2025-01-24:
Here is the debugger log for HS610:

```bash
08 E0 01 01 00 00 00 00 00 00 16 00 - Button 13 (ring) unpressed
08 E0 01 01 00 10 00 00 00 00 16 00 - Pressed
```

The ring is weird because it seems to affect the button states as well. I can only see the values going from 0-C.

Anticlockwise:

```bash
{ 08 F0 01 01 00 03 00 00 00 00 16 00 }, Delta:226.012ms, AuxButtons:[False False False False False False False False True True False False False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 04 00 00 00 00 16 00 }, Delta:111.997ms, AuxButtons:[False False False False False False False False False False True False False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 05 00 00 00 00 16 00 }, Delta:187.995ms, AuxButtons:[False False False False False False False False True False True False False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 06 00 00 00 00 16 00 }, Delta:75.668ms, AuxButtons:[False False False False False False False False False True True False False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 07 00 00 00 00 16 00 }, Delta:112.448ms, AuxButtons:[False False False False False False False False True True True False False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 08 00 00 00 00 16 00 }, Delta:113.871ms, AuxButtons:[False False False False False False False False False False False True False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 09 00 00 00 00 16 00 }, Delta:226.153ms, AuxButtons:[False False False False False False False False True False False True False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 00 00 00 00 00 16 00 }, Delta:563.995ms, AuxButtons:[False False False False False False False False False False False False False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 04 00 00 00 00 16 00 }, Delta:66151.909ms, AuxButtons:[False False False False False False False False False False True False False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 05 00 00 00 00 16 00 }, Delta:939.736ms, AuxButtons:[False False False False False False False False True False True False False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 06 00 00 00 00 16 00 }, Delta:490.296ms, AuxButtons:[False False False False False False False False False True True False False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 07 00 00 00 00 16 00 }, Delta:337.822ms, AuxButtons:[False False False False False False False False True True True False False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 08 00 00 00 00 16 00 }, Delta:263.894ms, AuxButtons:[False False False False False False False False False False False True False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 09 00 00 00 00 16 00 }, Delta:299.980ms, AuxButtons:[False False False False False False False False True False False True False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 0A 00 00 00 00 16 00 }, Delta:338.305ms, AuxButtons:[False False False False False False False False False True False True False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 0B 00 00 00 00 16 00 }, Delta:528.017ms, AuxButtons:[False False False False False False False False True True False True False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 0C 00 00 00 00 16 00 }, Delta:450.083ms, AuxButtons:[False False False False False False False False False False True True False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 01 00 00 00 00 16 00 }, Delta:376.663ms, AuxButtons:[False False False False False False False False True False False False False False False False False False False False], ReportType:UCLogicAuxReport
```

Clockwise (should be the reverse, right?

```bash
{ 08 F0 01 01 00 0A 00 00 00 00 16 00 }, Delta:639.972ms, AuxButtons:[False False False False False False False False False True False True False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 09 00 00 00 00 16 00 }, Delta:376.121ms, AuxButtons:[False False False False False False False False True False False True False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 08 00 00 00 00 16 00 }, Delta:338.175ms, AuxButtons:[False False False False False False False False False False False True False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 07 00 00 00 00 16 00 }, Delta:489.725ms, AuxButtons:[False False False False False False False False True True True False False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 06 00 00 00 00 16 00 }, Delta:338.030ms, AuxButtons:[False False False False False False False False False True True False False False False False False False False False], ReportType:UCLogicAuxReport
[... cut ...]
{ 08 F0 01 01 00 03 00 00 00 00 16 00 }, Delta:413.972ms, AuxButtons:[False False False False False False False False True True False False False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 02 00 00 00 00 16 00 }, Delta:490.067ms, AuxButtons:[False False False False False False False False False True False False False False False False False False False False], ReportType:UCLogicAuxReport
{ 08 F0 01 01 00 01 00 00 00 00 16 00 }, Delta:525.957ms, AuxButtons:[False False False False False False False False True False False False False False False False False False False False], ReportType:UCLogicAuxReport
```

Lucretia, 2025-01-26:
> Adding in my request for the HS610's wheel.

See my paste.

Tayasui-rainnya, 2025-03-09:
It's mine: XP-Pen Deco 02
Clockwise: [02 F0 00 00 00 00 00 01 00 00]
Counterclockwise: [02 F0 00 00 00 00 00 02 00 00]