OpenTabletDriver/OpenTabletDriver #1370: Cursor jitter on ctl-472

TurtleDuck33, 2021-08-24:
Hello, im trying to use this for the ctl-472, and i got everything working, exept for the fact that whenever the pen is in range of the tablet, it jitters from the place im aiming, to somewhere in the top left of my screen. Ive tried reading the FAQ and i cant find this anywhere. It took a while to get this working, please help

TurtleDuck33, 2021-08-24:
Update: i pressed "apply" in the bottom left and now its the bottom right? im confused

jjhhgg100123, 2021-08-25:
Hello! Please clarify your post a bit, and possibly provide video example and an exported diagnostics.

But the jitter you see/feel is the lack of smoothing, which can be added back in with plugins. It's the raw input of the tablet to the operating system, which naturally has "errors" due to physics.

Mrcubix, 2021-08-25:
Welcome to the world of raw input.

Compared to wacom, here in OTD, we don't forcefully enable bad predictive smoothing, It's all obtained through plugins that you can download via Plugin > Plugin manager, you can also change settings for each plugins.

What i'm saying is that the jitter you're talking about is in part from the tablet itself but mostly from you own hand, once again, you can use an antichatter plugin available in the plugin manager, or even a normal smoothing filter, All up to preference.

X9VoiD, 2021-08-25:
Will close, this is not an actual issue.

closed 2021-08-25 by X9VoiD