OpenTabletDriver/OpenTabletDriver #1375: Support Request: veikk S640 wont detect

mozurino, 2021-08-29:
# Support Request
 
new bought veikk S640 wont detect
 
[image]

## What I've already tried

i tried restarting after unistalling the original veikk drivers and switching usb ports open tablet wont detect the tablet
## Diagnostic Information

[diagnostic.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7071971/diagnostic.zip)

X9VoiD, 2021-08-29:
This is already fixed in OTD master.

closed 2021-08-29 by X9VoiD

mozurino, 2021-08-29:
> OTD master.

OTD master?

jjhhgg100123, 2021-08-30:
> 
> 
> > OTD master.
> 
> OTD master?

The latest build. Click on actions and download from there.