OpenTabletDriver/OpenTabletDriver #1378: Hover toggle

FloridaMen, 2021-08-29:
In version 0.23 of the now defunct tablet driver fork of hawku driver, there was a command called "RelativeDragMove", which when set to 1 would disable mouse input from the tablet unless the tip of the pen was pressed. This allows the mouse to be used as if it were an optical mouse, disabling movement when it is not touching a surface.

gonX, 2021-12-01:
[Kuuube's Hover Distance Limiter](https://github.com/Kuuuube/Hover_Distance_Limiter) does this when 0.6 releases.

jamesbt365, 2022-01-02:
> [Kuuube's Hover Distance Limiter](https://github.com/Kuuuube/Hover_Distance_Limiter) does this when 0.6 releases.

This has been updated completely for 0.6.0.0 and can be used just fine, closing as this has been added.

closed 2022-01-02 by jamesbt365