OpenTabletDriver/OpenTabletDriver #1381: Drag scroll by pressing supporting pen button

xillmera, 2021-08-30:
add chosable action to mouse properties at binding menu that allow : 
by pressing button and after changing cursor position (not only vertical but also in gorisontal plane) executed shifts on screen in required direction with some ratio|sensativity. Simular functionality can be found in another osce project "Inkscape", where by pressing "space" novigation on layer can be executed.

ziwzih, 2022-02-24:
yes need something like this

osuka, 2023-05-08:
one of my favourite functionalities, would love this back - it is in the x driver but no longer supported on wayland/libinput so it would be amazing to have it in opentabletdriver

axrolls-skoged, 2023-06-28:
yeah please add this. this would be extremely convenient.

xenio, 2024-01-06:
Is the scroll with button available? I am on MacOs and this is the only function I'm missing to switch from the wacom driver.

jamesbt365, 2024-01-06:
As this issue isnt closed, no.

SevenChords, 2024-06-28:
I second this as well, wanted to switch to tablet but still am required to use my mouse for quite a few applications as they don't support middle click + drag scrolling, just scrollwheel.

HollisMeynell, 2024-09-30:
This functionality is included in TableDriver, but it has not been updated for a long time. Will openTableDriver be supported?
[here](https://github.com/hawku/TabletDriver/blob/fd2185e0778b22839fb6b2f7598f529a8465ebaf/TabletDriverService/InputEmulator.cpp#L425)

Satanepigone, 2025-12-05:
> Worth noting that this is now indirectly supported via the plugin "[Scroll Bindings](https://github.com/Mrcubix/Scroll-Bindings/)" that you might be able to find in the Plugin Hub.

The "Scroll Bindings" you mentioned is very different from "Drag Scroll".
By Scroll binding, scroll is done with a fixed mount to a fixed direction each time when the button is pressed.
By Drag scroll, scrolling distance and direction is directly proportional to the movement of cursor/pen; like the scrolling on a smartphone screen.

Mrcubix, 2026-09-18:
Pen Scrolling is finally out of beta and should be available in the Plugin Manager sometimes soon.
It works accross platforms (except it doesn't perform as expected on Windows, because of its limitations).

Lot of fixes were applied since the release posted above, most of the issues should have disappeared.

About the Windows limitations, the behavior differ from an app to another.
On the Desktop, windows will only scroll in whichever axis has the highest scroll amount.
This means that horizontal & vertical scrolls events will conflict with each others.
There is no known way of fixing this issue right now.

You can get it from this release in the meantime : https://github.com/Mrcubix/Scroll-Bindings/releases/tag/1.1.0