OpenTabletDriver/OpenTabletDriver #1387: Add support for Wacom Cintiq Pro 13 (DTH-1320)

sp1ritCS, 2021-08-31:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer | Wacom |
| Product Name | DTH-1320 |
| Width           | 29.5cm |
| Height          | 16.5cm |
| Pressure Levels | 8192 |
| LPI             |

(libwacom db entry: https://github.com/linuxwacom/libwacom/blob/master/data/cintiq-pro-13.tablet)

## Diagnostic Information
```json
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:07.1/0000:09:00.3/usb5/5-3/5-3.4/5-3.4.1/5-3.4.1:1.0/0003:056A:034F.000F/hidraw/hidraw
11",
      "VendorID": 1386,
      "ProductID": 847,
      "ReleaseNumberBcd": 291,
      "CanOpen": true,
      "ReleaseNumber": "1.23",
      "ProductVersion": 291,
      "Manufacturer": "Wacom Co.,Ltd.",
      "ProductName": "Cintiq Pro 13",
      "SerialNumber": "8IQ0061003611",
      "MaxInputReportLength": 192,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 257
    },
[.. snip ..]
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:07.1/0000:09:00.3/usb5/5-3/5-3.4/5-3.4.2/5-3.4.2:1.0/0003:056A:0353.000E/hidraw/hidraw
10",
      "VendorID": 1386,
      "ProductID": 851,
      "ReleaseNumberBcd": 257,
      "CanOpen": true,
      "ReleaseNumber": "1.1",
      "ProductVersion": 257,
      "Manufacturer": "Wacom Co.,Ltd.      ",
      "ProductName": "Cintiq Pro 13 Touch",
      "SerialNumber": "",
      "MaxInputReportLength": 50,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 520
    },
```

I've gave creating my own config a shot, however failed to understand how to do so from that wiki entry. While it appears that using the TabletReportParser results in a "proper" X value  for the pointer, the Y value behaves like a vertical scanline (pointer moves quickly from top to bottom of screen). I suspect this being the parser confusing the y-value with some kind of synchronization counter.

gonX, 2021-12-06:
Thanks for the report. I think you're on the right track, you can try some of the other Wacom parsers. `IntuosV2ReportParser` is likely to be the correct one, but at worst we might have to make a new parser.

Posting OTD version used will help as well.

sp1ritCS, 2021-12-07:
Hmm, it seems to be able to parse something, but it isn't right :/.

both coordinates are wrong now, but the "pressure" the tablet debugger reads seems to correlate very well with the y-coordinate of my pen.
and the "hover distance" seems to correlate with pressure.

sp1ritCS, 2021-12-07:
oh, and yea I updated to 0.5.3

gonX, 2021-12-20:
If you can, try some of the other Wacom parsers. They're listed in [OpenTabletDriver.Configurations/Parsers/Wacom](https://github.com/OpenTabletDriver/OpenTabletDriver/tree/master/OpenTabletDriver.Configurations/Parsers/Wacom)

Alternatively, using a prerelease version (a recent version compiled from `master`) would allow us to record the full byte stream received from the data. This is done through the Tablet Debugger (Help -> Tablet Debugger in OTD's GUI), and then enabling the "Enable Data Recording" checkbox in the bottom right of the window.

Move the pen around in an X shape (be sure to note the direction, otherwise I suggest top left -> bottom right, top right -> bottom left, in that order), and try to always touch or never touch with the pen - it'll help simplify assigning bitfields and other values. The file will show up as `tablet-data.txt` in your OTD folder (`~/.config/OpenTabletDriver` on Linux), which if you upload here I can probably help make a parser for.

If you could annotate the tablet-data.txt file manually that would be amazing since it looks like the reports are quite large, otherwise I'll see what I can do. Please do not trim any lines unless you're trimming repeat starting/ending values.

jamesbt365, 2022-01-18:
Pretty sure there is another interface.

gonX, 2022-11-07:
We're still waiting for a tablet recording, or you trying the other parsers, before we can continue with this.

gonX, 2025-10-06:
Support added with #3063 (backport in #3092) or OpenTabletDriver v0.6.4.0.

closed 2025-10-06 by gonX