OpenTabletDriver/OpenTabletDriver #1389: Support Request:

TOXIC-MnM, 2021-09-03:
# Support Request
Wacom CTL-6100WL Plugged in says "No tablet found"

## What I've already tried
Searched online and havent found anything 

## Diagnostic Information
[Detect:Info]	Searching for tablet 'Gaomon 1060 Pro'
[Detect:Info]	Searching for tablet 'Gaomon M106K Pro'
[Detect:Info]	Searching for tablet 'Gaomon M106K'
[Detect:Info]	Searching for tablet 'Gaomon M10K'
[Detect:Info]	Searching for tablet 'Gaomon S56K'
[Detect:Info]	Searching for tablet 'Gaomon S620'
[Detect:Info]	Searching for tablet 'Huion 1060 Plus'
[Detect:Info]	Searching for tablet 'Huion G10T'
[Detect:Info]	Searching for tablet 'Huion GC610'
[Detect:Info]	Searching for tablet 'Huion H1060P'
[Detect:Info]	Searching for tablet 'Huion H1161'
[Detect:Info]	Searching for tablet 'Huion H430P'
[Detect:Info]	Searching for tablet 'Huion H610 Pro V2'
[Detect:Info]	Searching for tablet 'Huion H610 Pro'
[Detect:Info]	Searching for tablet 'Huion H640P'
[Detect:Info]	Searching for tablet 'Huion H950P'
[Detect:Info]	Searching for tablet 'Huion HC16'
[Detect:Info]	Searching for tablet 'Huion HS610'
[Detect:Info]	Searching for tablet 'Huion HS611'
[Detect:Info]	Searching for tablet 'Huion HS64'
[Detect:Info]	Searching for tablet 'Huion Kamvas 13'
[Detect:Info]	Searching for tablet 'Huion Kamvas 16'
[Detect:Info]	Searching for tablet 'Huion Kamvas Pro 13'
[Detect:Info]	Searching for tablet 'Huion New 1060 Plus'
[Detect:Info]	Searching for tablet 'Huion osu!tablet'
[Detect:Info]	Searching for tablet 'Huion Q11K V2'
[Detect:Info]	Searching for tablet 'Huion Q11K'
[Detect:Info]	Searching for tablet 'Huion Q620M'
[Detect:Info]	Searching for tablet 'Huion WH1409 V2'
[Detect:Info]	Searching for tablet 'VEIKK A15 Pro'
[Detect:Info]	Searching for tablet 'VEIKK A15'
[Detect:Info]	Searching for tablet 'VEIKK A30'
[Detect:Info]	Searching for tablet 'VEIKK A50'
[Detect:Info]	Searching for tablet 'VEIKK S640'
[Detect:Info]	Searching for tablet 'Wacom CTE-430'
[Detect:Info]	Searching for tablet 'Wacom CTE-440'
[Detect:Info]	Searching for tablet 'Wacom CTE-450'
[Detect:Info]	Searching for tablet 'Wacom CTE-460'
[Detect:Info]	Searching for tablet 'Wacom CTE-640'
[Detect:Info]	Searching for tablet 'Wacom CTE-650'
[Detect:Info]	Searching for tablet 'Wacom CTF-430'
[Detect:Info]	Searching for tablet 'Wacom CTH-460'
[Detect:Info]	Searching for tablet 'Wacom CTH-470'
[Detect:Info]	Searching for tablet 'Wacom CTH-480'
[Detect:Info]	Searching for tablet 'Wacom CTH-490'
[Detect:Info]	Searching for tablet 'Wacom CTH-670'
[Detect:Info]	Searching for tablet 'Wacom CTH-680'
[Detect:Info]	Searching for tablet 'Wacom CTH-690'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100WL'
[Detect:Info]	Searching for tablet 'Wacom CTL-460'
[Detect:Info]	Searching for tablet 'Wacom CTL-470'
[Detect:Info]	Searching for tablet 'Wacom CTL-471'
[Detect:Info]	Searching for tablet 'Wacom CTL-472'
[Detect:Info]	Searching for tablet 'Wacom CTL-480'
[Detect:Info]	Searching for tablet 'Wacom CTL-490'
[Detect:Info]	Searching for tablet 'Wacom CTL-6100WL'
[Detect:Info]	Searching for tablet 'Wacom CTL-671'
[Detect:Info]	Searching for tablet 'Wacom CTL-672'
[Detect:Info]	Searching for tablet 'Wacom CTL-680'
[Detect:Info]	Searching for tablet 'Wacom CTL-690'
[Detect:Info]	Searching for tablet 'Wacom FT-0405-U'
[Detect:Info]	Searching for tablet 'Wacom GD-0405-U'
[Detect:Info]	Searching for tablet 'Wacom GD-0608-U'
[Detect:Info]	Searching for tablet 'Wacom GD-0912-U'
[Detect:Info]	Searching for tablet 'Wacom GD-1212-U'
[Detect:Info]	Searching for tablet 'Wacom GD-1218-U'
[Detect:Info]	Searching for tablet 'Wacom MTE-450'
[Detect:Info]	Searching for tablet 'Wacom PTH-450'
[Detect:Info]	Searching for tablet 'Wacom PTH-451'
[Detect:Info]	Searching for tablet 'Wacom PTH-460'
[Detect:Info]	Searching for tablet 'Wacom PTH-650'
[Detect:Info]	Searching for tablet 'Wacom PTH-651'
[Detect:Info]	Searching for tablet 'Wacom PTH-660 Bluetooth'
[Detect:Info]	Searching for tablet 'Wacom PTH-660'
[Detect:Info]	Searching for tablet 'Wacom PTH-850'
[Detect:Info]	Searching for tablet 'Wacom PTH-851'
[Detect:Info]	Searching for tablet 'Wacom PTH-860'
[Detect:Info]	Searching for tablet 'Wacom PTK-1240'
[Detect:Info]	Searching for tablet 'Wacom PTK-440'
[Detect:Info]	Searching for tablet 'Wacom PTK-450'
[Detect:Info]	Searching for tablet 'Wacom PTK-540WL'
[Detect:Info]	Searching for tablet 'Wacom PTK-640'
[Detect:Info]	Searching for tablet 'Wacom PTK-650'
[Detect:Info]	Searching for tablet 'Wacom PTK-840'
[Detect:Info]	Searching for tablet 'Wacom PTZ-1230'
[Detect:Info]	Searching for tablet 'Wacom PTZ-1231W'
[Detect:Info]	Searching for tablet 'Wacom PTZ-430'
[Detect:Info]	Searching for tablet 'Wacom PTZ-431W'
[Detect:Info]	Searching for tablet 'Wacom PTZ-630'
[Detect:Info]	Searching for tablet 'Wacom PTZ-631W'
[Detect:Info]	Searching for tablet 'Wacom PTZ-930'
[Detect:Info]	Searching for tablet 'Wacom XD-0405-U'
[Detect:Info]	Searching for tablet 'Wacom XD-0608-U'
[Detect:Info]	Searching for tablet 'Wacom XD-0912-U'
[Detect:Info]	Searching for tablet 'Wacom XD-1212-U'
[Detect:Info]	Searching for tablet 'Wacom XD-1218-U'
[Detect:Info]	Searching for tablet 'XP-Pen Artist 12 Pro'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01 V2'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 02'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 03'
[Detect:Info]	Searching for tablet 'XP-Pen Deco Mini7'
[Detect:Info]	Searching for tablet 'XP-Pen Deco Pro Medium'
[Detect:Info]	Searching for tablet 'XP-Pen Deco Pro Small'
[Detect:Info]	Searching for tablet 'XP-Pen Star 03'
[Detect:Info]	Searching for tablet 'XP-Pen Star 06'
[Detect:Info]	Searching for tablet 'XP-Pen Star 06C'
[Detect:Info]	Searching for tablet 'XP-Pen Star G430'
[Detect:Info]	Searching for tablet 'XP-Pen Star G430S'
[Detect:Info]	Searching for tablet 'XP-Pen Star G540 Pro'
[Detect:Info]	Searching for tablet 'XP-Pen Star G540'
[Detect:Info]	Searching for tablet 'XP-Pen Star G640 V2'
[Detect:Info]	Searching for tablet 'XP-Pen Star G640'
[Detect:Info]	Searching for tablet 'XP-Pen Star G640S'
[Detect:Info]	Searching for tablet 'XP-Pen Star G960'
[Detect:Info]	Searching for tablet 'XP-Pen Star G960S Plus'
[Detect:Info]	No tablet found.
[Settings:Info]	Output mode: Absolute Mode
[Settings:Info]	Display area: [3840x2160@<5760, 1080>:0°],
[Settings:Info]	Tablet area: [152x95@<76, 47.5>:0°],
[Settings:Info]	Clipping: Enabled
[Settings:Info]	Ignoring reports outside area: Disabled
[Settings:Info]	Tip Binding: []@1%
[Settings:Info]	Pen Bindings: [0, ], [1, ]
[Settings:Info]	Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ], [6, ], [7, ]
[Settings:Info]	Driver is auto-enabled.

gonX, 2021-09-03:
Your tablet might be in Android mode. Searching closed issues on this bugtracker for 4100WL or 6100WL should help with debugging as well.

gonX, 2021-09-03:
Furthermore, if that doesn't solve it, you need to export diagnostics from OpenTabletDriver (through the Help menu) and upload the resulting file here.

TOXIC-MnM, 2021-09-03:
Tablet wasnt in android mode and i couldnt find anything in closed issues so here is the diagnostics

{
  "App Version": "OpenTabletDriver v0.5.3.3",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19042.0",
    "VersionString": "Microsoft Windows NT 10.0.19042.0"
  },
  "Environment Variables": {
    "FPS_BROWSER_USER_PROFILE_STRING": "Default",
    "APPDATA": "C:\\Users\\abc55\\AppData\\Roaming",
    "OneDrive": "C:\\Users\\abc55\\OneDrive",
    "CommonProgramFiles(x86)": "C:\\Program Files (x86)\\Common Files",
    "LOCALAPPDATA": "C:\\Users\\abc55\\AppData\\Local",
    "LOGONSERVER": "\\\\DESKTOP-JULPPMH",
    "ProgramFiles(x86)": "C:\\Program Files (x86)",
    "SystemDrive": "C:",
    "HOMEPATH": "\\Users\\abc55",
    "COMPUTERNAME": "DESKTOP-JULPPMH",
    "DriverData": "C:\\Windows\\System32\\Drivers\\DriverData",
    "NUMBER_OF_PROCESSORS": "16",
    "PROCESSOR_LEVEL": "23",
    "TEMP": "C:\\Users\\abc55\\AppData\\Local\\Temp",
    "USERPROFILE": "C:\\Users\\abc55",
    "CommonProgramW6432": "C:\\Program Files\\Common Files",
    "USERDOMAIN": "DESKTOP-JULPPMH",
    "PROCESSOR_REVISION": "7100",
    "PROCESSOR_ARCHITECTURE": "AMD64",
    "PUBLIC": "C:\\Users\\Public",
    "PATHEXT": ".COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC",
    "CommonProgramFiles": "C:\\Program Files\\Common Files",
    "VBOX_HWVIRTEX_IGNORE_SVM_IN_USE": "1",
    "TMP": "C:\\Users\\abc55\\AppData\\Local\\Temp",
    "ComSpec": "C:\\Windows\\system32\\cmd.exe",
    "windir": "C:\\Windows",
    "ProgramFiles": "C:\\Program Files",
    "ProgramW6432": "C:\\Program Files",
    "SystemRoot": "C:\\Windows",
    "ALLUSERSPROFILE": "C:\\ProgramData",
    "AMDRMPATH": "C:\\Program Files\\AMD\\RyzenMaster\\",
    "OneDriveConsumer": "C:\\Users\\abc55\\OneDrive",
    "USERDOMAIN_ROAMINGPROFILE": "DESKTOP-JULPPMH",
    "PROCESSOR_IDENTIFIER": "AMD64 Family 23 Model 113 Stepping 0, AuthenticAMD",
    "SESSIONNAME": "Console",
    "HOMEDRIVE": "C:",
    "FPS_BROWSER_APP_PROFILE_STRING": "Internet Explorer",
    "Path": "C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files (x86)\\Razer Chroma SDK\\bin;C:\\Program Files\\Razer Chroma SDK\\bin;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Windows\\System32\\OpenSSH\\;C:\\Users\\abc55\\AppData\\Local\\Microsoft\\WindowsApps;C:\\Program Files\\dotnet\\;C:\\Users\\abc55\\AppData\\Local\\Microsoft\\WindowsApps;",
    "ProgramData": "C:\\ProgramData",
    "PSModulePath": "C:\\Program Files\\WindowsPowerShell\\Modules;C:\\Windows\\system32\\WindowsPowerShell\\v1.0\\Modules",
    "USERNAME": "abc55",
    "OS": "Windows_NT"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#uvhid&col03#1&2d595ca7&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 38162,
      "ProductID": 38162,
      "CanOpen": true,
      "ReleaseNumberBcd": 5,
      "ReleaseNumber": "0.5",
      "ProductVersion": 5,
      "Manufacturer": "Unified Intents AB",
      "ProductName": "Unified Virtual HID",
      "SerialNumber": "123123123",
      "MaxInputReportLength": 2,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 0
    },
[... 6291 lines cut ...]
      "Time": "2021-09-03T12:49:47.2573178-07:00",
      "Group": "Detect",
      "Message": "No tablet found.",
      "Level": 1
    },
    {
      "Time": "2021-09-03T12:49:47.2573243-07:00",
      "Group": "IPC",
      "Message": "Disconnected from a client.",
      "Level": 0
    },
    {
      "Time": "2021-09-03T12:49:47.2573302-07:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "Level": 0
    },
    {
      "Time": "2021-09-03T12:49:47.257336-07:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "Level": 0
    }
  ]
}

closed 2021-09-04 by TOXIC-MnM