OpenTabletDriver/OpenTabletDriver #1390: Support request for serial Wacom tablet GD-0608-R

vancho1, 2021-09-04:
Hi,

I noticed you have support for the Wacom Intuos 2 tablets (GD-0608-U and GD-0405-U), but only the USB versions. I had my GD-0608-R working for many years with a hacked-together compatibility-mode installed Wacom driver but unfortunately at some point that driver broke. As far as I am aware, the -U and -R models are identical except for the -U is on USB and the -R is serial.

Kert, 2021-09-04:
Since no one replied yet
Does the tablet appear normally as a regular usb device when you use the converter? I am not sure but you can probably make a configuration file similar to other ones with correct VendorID/ProductID e.t.c.
I am suggesting this because probably you won't find another person here with a serial tablet

gonX, 2021-09-04:
We don't have serial support at the moment, but it would be cool to have. Pretty sure PRs are welcome, but it would likely need to work cross platform before being merged entirely - this is all assuming userspace has access to the serial port.

The device should theoretically work fine with a USB adapter as Kert pointed out - to determine this you need to post a diagnostic from OpenTabletDriver (Help -> Export Diagnostics) with the devices plugged in.

gonX, 2021-09-04:
Reading some other contributors thoughts on the Discord made me realize a USB adapter would likely just present itself as a serial port, which the driver does not support yet. A diagnostics log from OTD would answer whether this is the case.

If you or anyone else plans to implement serial support, please join our Discord first, since it requires quite extensive backend changes.

vancho1, 2021-09-04:
Taking a look at the device manager in Windows, I believe you're right - my adapter shows up as a COM (ie serial) port, and furthermore the tablet itself only shows up in my HID section when I plug it in to the physical serial port. Unfortunate! I'll hop in your discord at some point; I'm not really sure what I'd need to do to add serial support to the program but if I have time I can try and figure it out.

jamesbt365, 2025-01-24:
Closing until a backend for OTD to support this is added, which is currently out of scope but PRs are welcome.

closed 2025-01-24 by jamesbt365