OpenTabletDriver/OpenTabletDriver #1396: Can't detect tablet. Wacom 472

SWAGEDGAR, 2021-09-06:
The tablet got detected on OTD when Wacom driver was still installed but Wacom Driver took control of the tablet (area, filter, etc) so OTD changed nothing so I decided to uninstall Wacom Driver but now it won't get detected on OTD.

InfinityGhost, 2021-09-06:
How did you uninstall the Wacom driver? If you just deleted the files for the driver you'll need to reinstall it then use add or remove programs to remove it.

SWAGEDGAR, 2021-09-06:
> How did you uninstall the Wacom driver? If you just deleted the files for the driver you'll need to reinstall it then use add or remove programs to remove it.

I used a program that uninstalls stuff

jamesbt365, 2021-09-06:
Can you export daignosis, go to help > export daig and send the file here for me please.

SWAGEDGAR, 2021-09-06:
> Can you export daignosis, go to help > export daig and send the file here for me please.
Didn't let me sent through here so I put it in a google drive 
https://drive.google.com/drive/folders/1tIUJQ22ZsGLDHJTr6ofz6P-j2VDwQ8-W?usp=sharing

jamesbt365, 2021-09-06:
`     "Time": "2021-09-06T17:50:38.466849-04:00",
      "Group": "Detect",
      "Message": "Using device 'CTL-472'.",
      "Level": 0
    },
    {
      "Time": "2021-09-06T17:50:38.466863-04:00",
      "Group": "Detect",
      "Message": "Using report parser type 'OpenTabletDriver.Tablet.TabletReportParser'.",
      "Level": 0
    },
    {
      "Time": "2021-09-06T17:50:38.466878-04:00",
      "Group": "Detect",
      "Message": "Device path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/EHC2@1A,7/EHC2@fa000000/PRT1@fa100000/IOUSBHostDevice@fa100000/AppleUSB20InternalHub@fa100000/AppleUSB20HubPort@fa130000/CTL-472@fa130000/IOUSBHostInterface@0/IOUSBHostHIDDevice@fa130000,0",
      "Level": 0
    },
    {
      "Time": "2021-09-06T17:50:38.466893-04:00",
      "Group": "Device",
      "Message": "Set tablet feature: 02-02",
      "Level": 0
    },
    {
      "Time": "2021-09-06T17:50:38.466907-04:00",
      "Group": "Detect",
      "Message": "Found tablet 'Wacom CTL-472'",
      "Level": 1
    },
    {
      "Time": "2021-09-06T17:50:38.466922-04:00",
      "Group": "Detect",
      "Message": "Using device 'CTL-472'.",
      "Level": 0
    },
    {
      "Time": "2021-09-06T17:50:38.46694-04:00",
      "Group": "Detect",
      "Message": "Using auxiliary report parser type 'AuxReportParser'.",
      "Level": 0
    },
    {
      "Time": "2021-09-06T17:50:38.466955-04:00",
      "Group": "Detect",
      "Message": "Device path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/EHC2@1A,7/EHC2@fa000000/PRT1@fa100000/IOUSBHostDevice@fa100000/AppleUSB20InternalHub@fa100000/AppleUSB20HubPort@fa130000/CTL-472@fa130000/IOUSBHostInterface@1/IOUSBHostHIDDevice@fa130000,1",
      "Level": 0
    },`

Next person who sees this, this is what i see.

SWAGEDGAR, 2021-09-06:
> ` "Time": "2021-09-06T17:50:38.466849-04:00", "Group": "Detect", "Message": "Using device 'CTL-472'.", "Level": 0 }, { "Time": "2021-09-06T17:50:38.466863-04:00", "Group": "Detect", "Message": [...]

It works now. I didn't touch anything except the diagnostic thing

closed 2021-09-06 by SWAGEDGAR