OpenTabletDriver/OpenTabletDriver #1401: Valid inputs being dropped - Hover distance and area issues

Kuuuube, 2021-09-07:
## Description

Many valid inputs are being ignored leading certain tablets to have their input cut off before it should be. This can lead to lowered hover distance, and certain parts of areas (specifically some parts of the area deadzones on wacom tablets) being unreachable or having input cut off too soon.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| OpenTabletDriver Version | 0.6.0 Pre-release
| Tablet                   | Tested on CTL-480 but many others likely affected

jamesbt365, 2021-09-07:
How would i check for this on my xp pen G640 Rev a?

Kuuuube, 2021-09-07:
> 
> 
> How would i check for this on my xp pen G640 Rev a?

Go to `tablets > tablet debugger` and check if theres any points where the `Raw Tablet Data` box still shows valid position bytes sent but there is no position shown in the `Tablet Report` box. 
Usually this should be around the edges of the tablet or when hovering nearly out of range.

jamesbt365, 2021-09-07:
When moving out on one of the sides of my tablet and come back into range my cursor gets stuck even though it still moves on the tablet debugger, it says there for about a second unless i manually replace my pen. This only happens occasionally, however i have found no dead zones or spots on my G640 Rev A.

I don't think the hover is high enough to really be cut off but i haven't really done much testing on that regard. Going to do more testing when i come back from school tomorrow but i`m unsure if this is Wacom only.

InfinityGhost, 2021-12-05:
This issue really needs to be narrowed down to track individual cases, right now we have no idea if this is resolved.

Kuuuube, 2021-12-05:
I don't have time right now to test this on all my tablets and that will probably have to wait a week or so but I can give you this: The Wacom CTL-480 reports get ignored when it has `D0` in the second byte which cuts off a lot of the hover.

gonX, 2021-12-06:
Likely fixed by #1698

PR #1698 (2021-12-13): Fix IntuosReportParser to read out NearProximity==false reports (hover range fix)
This seems to be an off-by-one error. Should fix issues with the following tablets having reduced hover range, which is a regression from 0.5.3.

Closes #1401

Needs testing:
- [ ] CTE-430
- [ ] CTE-440
- [ ] CTE-450
- [ ] CTE-460
- [ ] CTE-630
- [ ] CTE-640
- [ ] CTE-650
- [ ] CTF-430
- [ ] CTH-460
- [ ] CTH-470
- [ ] CTH-661
- [ ] CTH-670
- [ ] CTL-671
- [ ] CTL-672
- [ ] CTL-680
- [ ] ET-0405A-U
- [ ] FT-0405-U
- [x] CTH-461
- [x] CTH-480
- [x] CTH-680
- [x] CTL-460
- [x] CTL-470
- [x] CTL-471
- [x] CTL-472
-  [... cut]
PR #1679 (2021-12-13): Update configurations and parsers
- Closes #1101
- Closes #1282
- Closes #1338
- Closes #1401
- Closes #1435
- Closes #1448
- Closes #1500
- Closes #1545
closed 2021-12-13 by InfinityGhost