OpenTabletDriver/OpenTabletDriver #1402: Filters are not refreshed when new plugins are added

Kuuuube, 2021-09-07:
## Description

Previously after installing a new plugin the filters tab would be refreshed and the plugin would be shown there. That seems to no longer be the case. 
I have only tested this with manually installed plugins and cannot yet verify if installing through the plugin repo has this issue as well.
Restarting OTD causes the plugin to show and it does work normally.

Tested using an updated version of Kuuube's Chatter Exterminator. [Download Here](https://cdn.discordapp.com/attachments/628651971267788800/884592400402632744/Kuuubes_CHATTER_EXTERMINATOR.dll)
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 10
| OpenTabletDriver Version | 0.6.0 Pre-release

jamesbt365, 2021-09-07:
I can confirm that this is not a one of thing and that it has happened multiple times for me.

Kuuuube, 2021-09-07:
This appears to happen only when installing a plugin when there are no plugins currently loaded.

Two more plugins to help with testing this:
[Hover Distance Limiter](https://cdn.discordapp.com/attachments/628651971267788800/884604847805042788/Hover_Distance_Limiter.dll)
[Tablet Debounce](https://cdn.discordapp.com/attachments/628651971267788800/884604847981223956/Tablet_Debounce.dll)

Kuuuube, 2022-01-02:
This is not fixed. Tested with 0.6.0rc2 on windows 10.

Here's Circular Area for 0.6.0 for testing: [2.0.0_release_Circular_Area.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7799849/2.0.0_release_Circular_Area.zip)

Kuuuube, 2022-01-02:
0.6.0rc2 windows release was built off the wrong commit. This is fixed on the corrected 0.6.0rc2.

PR #1809 (2022-01-02): Refresh PluginSettingStoreCollectionEditor's Content when assemblies change
This allows plugins to be displayed in the Filters or Tools tab when they are first added.
Changed from `sourceSelector.DataStore` to getting child types directly to avoid depending on a specific execution order.
`TypeListBox` was used as the main point of reference.

Closes #1402

My understanding is that the plugin list (when it's there) is refreshing using this, so I ported it over to `PluginSettingStoreCollectionEditor` to also refresh the tabs: https://github.com/OpenTabletDriver/OpenTabletDriver/blob/d75802318360 [... cut]
commit ebe098de (2022-01-02): Refresh PluginSettingStoreCollectionEditor's Content when assemblies change

This allows plugins to displayed in the Filters or Tools tab when they are first added.
Changed sourceSelector.DataStore to get child types directly to avoid depending on a specific execution order.
TypeListBox was used as the main point of reference.

Closes #1402
closed 2022-01-02 by Kuuuube