OpenTabletDriver/OpenTabletDriver #1403: G430S has lower report rate on 0.6.0

coinworld, 2021-09-07:
## Description

G430S (and possibly other tablets) has lower report rate on 0.6.0, and cursor movement is significantly slower.
This doesn't happen on CTL-472, so I think it's because of some changes between these two versions.

---

[image]
^ 0.5.3.3

[image]
^ 0.6.0 pre-release

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | openSUSE Tumbleweed
| OpenTabletDriver Version | 0.6.0 pre-release on discord
| Tablet                   | XP-Pen G430S

iotku, 2021-09-08:
Can confirm, troubleshot this a fair bit on discord in #help-linux but didn't come to any meaningful conclusion. (Was planned to create an issue but I don't think one has been made prior to this.)

Same issue on my system.

Name | Value
-- | --
Operating System | Arch Linux
OpenTabletDriver Version | 0.6.0 pre-release on discord (and previously on master via opentabletdriver-git AUR Package)
Tablet | XP-Pen G430S

iotku, 2021-09-08:
Of slight possible note v0.6.0.0 detects my tablet as "XP-Pen Star 430S", while v0.5.3.3 detects my tablet as "XP-Pen Star 430" (No S at the end) so it could potentially be using a different code-path? (or was just renamed)

closed 2021-10-12 by coinworld