OpenTabletDriver/OpenTabletDriver #1404: Support Request: Huion 1060Pro+ is not detected, but drawing works, Short keys dont.

Pisseschwitzer, 2021-09-07:
# Support Request
 
Huion 1060Pro+ is not detected, but drawing works, Short keys dont.

OpenTabletDriver got installed on a fresh set up Windows 10, no Huion driver was installed before.
The Key bindings are defaultly set to ctrl+x ctrl+v and other few important ones that i dont like to change in function. 
In the driver i like to set those keys to the num block number keys, so i can map them without conflict in Krita.

I assume its just not compatible with the driver, as its not on the compatibility list,

I like to help getting it compatible.

## What I've already tried
 I tried reinstalling the driver manually as suggested on this board, the names are different on my system, so i couldnt fix it.
[image]
I got 2 undetected devices, they stay when i unplug the Tablet and the Pen

## Diagnostic Information

[huion 1060pro+ not detected.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7120557/huion.1060pro%2B.not.detected.txt)

Greetings, Alex

jamesbt365, 2021-09-07:
Have you used winusb?

Follow the instructions here:

https://github.com/X9VoiD/WinUSBCleanup/releases/tag/v1.0

Pisseschwitzer, 2021-09-07:
Yes i have used it, but it dint found anything to remove.

"Huion/Gaomon Cleanup Script

No Gaomon/Huion driver detected

Press any key to exit..."

jamesbt365, 2021-09-07:
Ah i didnt realise you said that it isnt on the compatibility list. From what i can see i dont think a config can be made as i dont see a 10 0 0. However I am unsure and somebody is going to have to look at it.

Mrcubix, 2021-09-07:
Your tablet's firmware make it impossible to be detected and made compatible by OTD, other tablet with a different firmware might work if they don't entirely rely on WinUSB. if you bought this tablet recently, you might consider getting a refund and get a gaomon S620 which is known compatible no matter the firmware OR you can wait for a build where WinUSB is properly implemented (which was merged early i believe?) and where ghost fixed access to tablet property (necessary for some plugins to work)

Pisseschwitzer, 2021-09-07:
Sadly i bought it a couple years back, installed the huion driver, messed up my system with them, and since then it was laying around in my room.
Until yesterday i only tried the tablet by the huion installation routine, wchis says first install the driver and then plug in the tablet.
Which makes me question, as i can draw with it, does it runs on windows default drivers now?

I will wait for the WinUSB fix and then try again.

Mrcubix, 2021-09-07:
you tablet could be using it's plug and play functionality and is running using windows ink but i don't know if it support pressure

Pisseschwitzer, 2021-09-07:
I now remember having tried the tablet without driver once, there it didnt change the preassure, but i cant remember if i tested it in krita or just with gimp.

As it now works with preassure, i finally can use it. And if the ShortKey mapping works some day in future that would be great.

Thanks for the help.

Mrcubix, 2021-09-09:
Join the [discord server](https://discord.gg/P3GYdUS) when you have time, WinUSB support has been added in pre-release and it will be faster to get it compatible if you join

Pisseschwitzer, 2021-09-10:
Allright, im going to write in help-windows

gonX, 2021-12-06:
Assuming resolved given the age of the issue

closed 2021-12-06 by gonX