OpenTabletDriver/OpenTabletDriver #1407: Current 0.6.0 Pre-release Plugin API does not support some features from 0.5.3.3

Kuuuube, 2021-09-08:
## Description

Any plugin that relied on reading tablet/monitor specific properties currently cannot be updated for 0.6.0. These properties are things such as: tablet size specifications, tablet area settings, and monitor area settings.

There may be some other things missing but as far as I know these are the only things that cause breakages in current plugins.

The following plugins on the current plugin repo cannot be updated:
TabletDriverFilters
Radial Follow Smoothing
Windows Ink
Circular Area
Tablet Area Randomizer
Gess1t's Area Randomizer
## System Information:

| Name             | Value |
| ---------------- | ----- |
| OpenTabletDriver Version | 0.6.0 Pre-release

25pwn, 2021-09-08:
More plugins also get `can't be loaded and is likely out of date`:

```
[RootHub:Debug] Initialized internal child hubs: WinUSBRootHub, HidSharpDeviceRootHub
[Plugin:Debug]  Loading plugin 'AbstractQbit's Radial Follow Smoothing'
[Plugin:Warning]        Plugin 'RadialFollow, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Debug]  Loading plugin 'AudioToggle'
[Plugin:Warning]        Plugin 'AudioToggle, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Debug]  Loading plugin 'BezierInterpolator'
[Plugin:Warning]        Plugin 'BezierInterpolator, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Debug]  Loading plugin 'Circular Area'
[Plugin:Warning]        Plugin 'Circular_Area, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Debug]  Loading plugin 'DevocubFilters'
[Plugin:Warning]        Plugin 'DevocubFilters, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Debug]  Loading plugin 'Gess1t's Area Randomizer'
[Plugin:Warning]        Plugin 'Gess1t's Area Randomizer, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Debug]  Loading plugin 'Gess1t's Proxy API'
[Plugin:Debug]  Loading plugin 'HawkuFilters'
[Plugin:Warning]        Plugin 'HawkuFilters, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Debug]  Loading plugin 'Kuuube's CHATTER EXTERMINATOR'
[Plugin:Warning]        Plugin 'Kuuube's_CHATTER_EXTERMINATOR, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Debug]  Loading plugin 'MeL'
[Plugin:Warning]        Plugin 'MeL, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Debug]  Loading plugin 'OemKill'
[Plugin:Debug]  Loading plugin 'Precision Control'
[Plugin:Warning]        Plugin 'PrecisionControl, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Debug]  Loading plugin 'Reconstructor'
[Plugin:Warning]        Plugin 'Reconstructor, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Debug]  Loading plugin 'ScriptRunner'
[Plugin:Warning]        Plugin 'ScriptRunner, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Debug]  Loading plugin 'SpringInterpolator'
[Plugin:Warning]        Plugin 'SpringInterpolator, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Debug]  Loading plugin 'Tablet Area Randomizer'
[Plugin:Warning]        Plugin 'Tablet_Area_Randomizer, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Debug]  Loading plugin 'TouchEmu'
[Plugin:Warning]        Plugin 'TouchEmu, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Debug]  Loading plugin 'VMultiMode'
[Plugin:Warning]        Plugin 'VMulti, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Warning]        Plugin 'VMultiMode, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Debug]  Loading plugin 'Windows Ink'
[Plugin:Warning]        Plugin 'VMulti, Version=1.0.0.0' can't be loaded and is likely out of date.
[Plugin:Warning]        Plugin 'WindowsInk, Version=1.0.0.0' can't be loaded and is likely out of date.
[Detect:Info]   Searching for tablet 'Wacom CTL-472'
[Device:Debug]  Initializing device 'CTL-472' \\?\hid#vid_056a&pid_037a&mi_00&col02#7&1c35bcba&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
[Device:Debug]  Using report parser type 'OpenTabletDriver.Vendors.Wacom.Intuos.IntuosReportParser'
[Device:Debug]  Set device feature: 02-02
[Detect:Info]   Found tablet 'Wacom CTL-472'
[Device:Debug]  Initializing device 'CTL-472' \\?\hid#vid_056a&pid_037a&mi_01#7&33f948ff&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
[Device:Debug]  Using report parser type 'OpenTabletDriver.Vendors.Wacom.Wacom64bAuxReportParser'
[Display:Debug] DPI Awareness enabled
[Wacom CTL-472:Info]    Output mode: Absolute Mode
[Wacom CTL-472:Info]    Display area: [3840x2160@<960, 540>:0°],
[Wacom CTL-472:Info]    Tablet area: [51.2x28.8@<121.75, 58>:-90°],
[Wacom CTL-472:Info]    Clipping: Enabled
[Wacom CTL-472:Info]    Ignoring reports outside area: Enabled
[Wacom CTL-472:Info]    Tip Binding: [Mouse Button Binding: Left]@0%
[Wacom CTL-472:Info]    Pen Bindings: Mouse Button Binding: Right, Mouse Button Binding: Middle
[Settings:Info] Driver is enabled.
```

X9VoiD, 2021-09-08:
@25pwn you should use the dev builds of plugins when available, because it is exactly what it says, OTD v0.6.0 does not support plugins made for OTD 0.5.x

jamesbt365, 2021-10-03:
There is a new radial flow build that runs of pixels now