OpenTabletDriver/OpenTabletDriver #1408: Plugging in and unplugging multiple tablets can cause OpenTabletDriver.UX to freeze

Kuuuube, 2021-09-08:
## Description

I was able to trigger this bug and freeze OTD UX in these two ways:

```
start with Tablet 1 plugged in or plug it in after OTD starts
Tablet 2 plugged in
Tablet 2 unplugged
Tablet 2 plugged in
```
```
start with Tablet 1 plugged in or plug it in after OTD starts
Tablet 2 plugged in
Tablet 1 unplugged
Tablet 2 plugged in
```
I have not tested with more than two tablets but there are likely many combinations that could cause this bug.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 10
| OpenTabletDriver Version | 0.6.0 Pre-release

gonX, 2021-12-01:
Is this #1627 ?

Kuuuube, 2021-12-01:
At the time this was tested no stack traces were thrown and there were no crashes by doing the actions mentioned only a gui freeze. I haven't tested this on current master though.

InfinityGhost, 2021-12-17:
This needs testing, likely was resolved alongside #1627

Kuuuube, 2021-12-23:
Tested on 2f9f73eb153b7dfe36bd998e0c49419b26247be8 cannot reproduce this issue anymore.

closed 2021-12-23 by Kuuuube