OpenTabletDriver/OpenTabletDriver #1410: Plugging in and unplugging tablets can result in a blank settings display in the UX

Kuuuube, 2021-09-08:
## Description

Starting OTD with a tablet plugged in then unplugging it and plugging in another tablet will cause the UX to not automatically switch to the currently plugged in tablet and instead display blank settings for a tablet with a blank name.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 10
| OpenTabletDriver Version | 0.6.0 Pre-release

InfinityGhost, 2021-12-15:
Unable to reproduce as of ecc3218c49b5ca5796c434b994f730cd81d238e7

AkiSakurai, 2021-12-16:
I can reproduce the problem on MacOS by unplugging and plugging in the same tablet.

The problem is caused by multiple DevicesChanged Events.

There are two issues.

1. This async procedure can be called multiple times concurrently, causing random exceptions or duplicated devices.

https://github.com/OpenTabletDriver/OpenTabletDriver/blob/7dc0cd01fc2e8583b14982ee811d0e14d4180f5c/OpenTabletDriver.Daemon/DriverDaemon.cs#L39

2.  The method Open can return null, and causing null pointer exception when Main is called. 

https://github.com/OpenTabletDriver/OpenTabletDriver/blob/7dc0cd01fc2e8583b14982ee811d0e14d4180f5c/OpenTabletDriver/Devices/DeviceReader.cs#L83
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/7dc0cd01fc2e8583b14982ee811d0e14d4180f5c/OpenTabletDriver/Devices/DeviceReader.cs#L109

I am not 100% OP`s problem is caused by these issues, as the OS is different.

Kuuuube, 2021-12-23:
Tested on 2f9f73eb153b7dfe36bd998e0c49419b26247be8 on windows 10 cannot reproduce this issue anymore. Will leave this open though.

InfinityGhost, 2021-12-23:
> I can reproduce the problem on MacOS by unplugging and plugging in the same tablet.
>
> The problem is caused by multiple DevicesChanged Events.
>
> There are two issues.
>
>     1. This async procedure can [...]

This should be tracked as it's own issue. It should be throwing some device exception that says the endpoint failed to open.

InfinityGhost, 2021-12-23:
> Tested on [2f9f73e](https://github.com/OpenTabletDriver/OpenTabletDriver/commit/2f9f73eb153b7dfe36bd998e0c49419b26247be8) on windows 10 cannot reproduce this issue anymore. Will leave this open though.

Closing as this was the original issue source.

closed 2021-12-23 by InfinityGhost