OpenTabletDriver/OpenTabletDriver #1411: Starting OTD then plugging in a tablet with no saved settings causes a blank screen

Kuuuube, 2021-09-08:
## Description

Starting the UX with no tablets plugged in then plugging in a tablet that has no saved settings will cause a blank screen with no info on it. It does not have blank settings it just has nothing except the tablet switcher dropdown. 

Switching manually to the plugged in tablet does make settings show properly.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 10
| OpenTabletDriver Version | 0.6.0 Pre-release

InfinityGhost, 2021-12-16:
Unable to reproduce as of ecc3218c49b5ca5796c434b994f730cd81d238e7

Kuuuube, 2021-12-23:
Tested on 2f9f73eb153b7dfe36bd998e0c49419b26247be8 cannot reproduce this issue anymore.

closed 2021-12-23 by Kuuuube