OpenTabletDriver/OpenTabletDriver #1418: Wacom CTH-661 is not available on MacOS

cirru05, 2021-09-09:
## Description

**The tablet Wacom CTH-661 is not on the list of the detectable tablets in the application, and it does not have a configuration file by default on Mac.** But it is listed in the Supported Tablets file: [TABLETS.md](https://github.com/OpenTabletDriver/OpenTabletDriver/blob/master/TABLETS.md).
Here is the log of tablets which it tried to detect: [Tablet_search_log.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7139027/Tablet_search_log.txt). It does not contain the CTH-661.

Nor the configuration folder (the one which is inside the app):
`-rw-r--r--@ 1 boldizsar  staff  1273 Jul 22 19:43 CTE-430.json`
`-rw-r--r--@ 1 boldizsar  staff  1588 Jul 22 19:43 CTE-440.json`
`-rw-r--r--@ 1 boldizsar  staff   692 Jul 22 19:43 CTE-450.json`
`-rw-r--r--@ 1 boldizsar  staff  1275 Jul 22 19:43 CTE-460.json`
`-rw-r--r--@ 1 boldizsar  staff  1329 Jul 22 19:43 CTE-640.json`
`-rw-r--r--@ 1 boldizsar  staff   723 Jul 22 19:43 CTE-650.json`
`-rw-r--r--@ 1 boldizsar  staff   717 Jul 22 19:43 CTF-430.json`
`-rw-r--r--@ 1 boldizsar  staff  2591 Jul 22 19:43 CTH-460.json`
`-rw-r--r--@ 1 boldizsar  staff  1615 Jul 22 19:43 CTH-470.json`
`-rw-r--r--@ 1 boldizsar  staff  1615 Jul 22 19:43 CTH-480.json`
`-rw-r--r--@ 1 boldizsar  staff  1633 Jul 22 19:43 CTH-490.json`
`-rw-r--r--@ 1 boldizsar  staff  1621 Jul 22 19:43 CTH-670.json`
`-rw-r--r--@ 1 boldizsar  staff  1619 Jul 22 19:43 CTH-680.json`
`-rw-r--r--@ 1 boldizsar  staff  1637 Jul 22 19:43 CTH-690.json`
`-rw-r--r--@ 1 boldizsar  staff  1298 Jul 22 19:43 CTL-4100.json`
`-rw-r--r--@ 1 boldizsar  staff  2481 Jul 22 19:43 CTL-4100WL.json`
`-rw-r--r--@ 1 boldizsar  staff  1599 Jul 22 19:43 CTL-460.json`
`-rw-r--r--@ 1 boldizsar  staff  1600 Jul 22 19:43 CTL-470.json`
`-rw-r--r--@ 1 boldizsar  staff  1600 Jul 22 19:43 CTL-471.json`
`-rw-r--r--@ 1 boldizsar  staff  1600 Jul 22 19:43 CTL-472.json`
`-rw-r--r--@ 1 boldizsar  staff  1615 Jul 22 19:43 CTL-480.json`
`-rw-r--r--@ 1 boldizsar  staff  1633 Jul 22 19:43 CTL-490.json`
`-rw-r--r--@ 1 boldizsar  staff  1296 Jul 22 19:43 CTL-6100WL.json`
`-rw-r--r--@ 1 boldizsar  staff  1606 Jul 22 19:43 CTL-671.json`
`-rw-r--r--@ 1 boldizsar  staff  1604 Jul 22 19:43 CTL-672.json`
`-rw-r--r--@ 1 boldizsar  staff  1619 Jul 22 19:43 CTL-680.json`
`-rw-r--r--@ 1 boldizsar  staff  1637 Jul 22 19:43 CTL-690.json`
`-rw-r--r--@ 1 boldizsar  staff  1274 Jul 22 19:43 FT-0405-U.json`
`-rw-r--r--@ 1 boldizsar  staff  1353 Jul 22 19:43 GD-0405-U.json`
`-rw-r--r--@ 1 boldizsar  staff  1303 Jul 22 19:43 GD-0608-U.json`
`-rw-r--r--@ 1 boldizsar  staff  1303 Jul 22 19:43 GD-0912-U.json`
`-rw-r--r--@ 1 boldizsar  staff  1303 Jul 22 19:43 GD-1212-U.json`
`-rw-r--r--@ 1 boldizsar  staff  1303 Jul 22 19:43 GD-1218-U.json`
`-rw-r--r--@ 1 boldizsar  staff  1273 Jul 22 19:43 MTE-450.json`
`-rw-r--r--@ 1 boldizsar  staff  1630 Jul 22 19:43 PTH-450.json`
`-rw-r--r--@ 1 boldizsar  staff  1633 Jul 22 19:43 PTH-451.json`
`-rw-r--r--@ 1 boldizsar  staff  1658 Jul 22 19:43 PTH-460.json`
`-rw-r--r--@ 1 boldizsar  staff  1698 Jul 22 19:43 PTH-650.json`
`-rw-r--r--@ 1 boldizsar  staff  1639 Jul 22 19:43 PTH-651.json`
`-rw-r--r--@ 1 boldizsar  staff   705 Jul 22 19:43 PTH-660 Bluetooth.json`
`-rw-r--r--@ 1 boldizsar  staff  1032 Jul 22 19:43 PTH-660.json`
`-rw-r--r--@ 1 boldizsar  staff  1636 Jul 22 19:43 PTH-850.json`
`-rw-r--r--@ 1 boldizsar  staff  1639 Jul 22 19:43 PTH-851.json`
`-rw-r--r--@ 1 boldizsar  staff  1628 Jul 22 19:43 PTH-860.json`
`-rw-r--r--@ 1 boldizsar  staff  1358 Jul 22 19:43 PTK-1240.json`
`-rw-r--r--@ 1 boldizsar  staff  1310 Jul 22 19:43 PTK-440.json`
`-rw-r--r--@ 1 boldizsar  staff  1297 Jul 22 19:43 PTK-450.json`
`-rw-r--r--@ 1 boldizsar  staff  1357 Jul 22 19:43 PTK-540WL.json`
`-rw-r--r--@ 1 boldizsar  staff  1355 Jul 22 19:43 PTK-640.json`
`-rw-r--r--@ 1 boldizsar  staff  1303 Jul 22 19:43 PTK-650.json`
`-rw-r--r--@ 1 boldizsar  staff  1305 Jul 22 19:43 PTK-840.json`
`-rw-r--r--@ 1 boldizsar  staff  1305 Jul 22 19:43 PTZ-1230.json`
`-rw-r--r--@ 1 boldizsar  staff  1308 Jul 22 19:43 PTZ-1231W.json`
`-rw-r--r--@ 1 boldizsar  staff  1304 Jul 22 19:43 PTZ-430.json`
`-rw-r--r--@ 1 boldizsar  staff  1309 Jul 22 19:43 PTZ-431W.json`
`-rw-r--r--@ 1 boldizsar  staff  1355 Jul 22 19:43 PTZ-630.json`
`-rw-r--r--@ 1 boldizsar  staff  1309 Jul 22 19:43 PTZ-631W.json`
`-rw-r--r--@ 1 boldizsar  staff  1355 Jul 22 19:43 PTZ-930.json`
`-rw-r--r--@ 1 boldizsar  staff  1301 Jul 22 19:43 XD-0405-U.json`
`-rw-r--r--@ 1 boldizsar  staff  1301 Jul 22 19:43 XD-0608-U.json`
`-rw-r--r--@ 1 boldizsar  staff  1301 Jul 22 19:43 XD-0912-U.json`
`-rw-r--r--@ 1 boldizsar  staff  1301 Jul 22 19:43 XD-1212-U.json`
`-rw-r--r--@ 1 boldizsar  staff  1301 Jul 22 19:43 XD-1218-U.json`
 
I tried older versions as well, because I saw a similar report on [reddit](https://www.reddit.com/r/wacom/comments/e0w7q0/why_does_my_wacom_bamboo_fun_cth_661_tablet_not/), but it did not fix anything. And before you ask I am running on High Sierra (10.13.6). I uninstalled everything that could possibly block the tablet from getting recognised by the software (i.e.: I had the original Wacom driver installed, as well as an input modification software called USBOverride, and uninstalled both, but nothing happened).
Also it is possible that it is just me being stupid, if yes, than tell me please.
Any information is welcome, and sorry if I sounded like I do not respect the software, but I do.
Help me as soon as possible, so I can click circles again!

X9VoiD, 2021-09-09:
Use OTD pre-release version from OTD's Discord server. Current release do not have support for CTH-661.

cirru05, 2021-09-09:
Thank you, it works!

closed 2021-09-09 by cirru05