OpenTabletDriver/OpenTabletDriver #1427: Half of tablet area doesn't take up width

blobotic, 2021-09-11:
## Description
When using OTB with two monitors and an XP-PEN Deco 01 V2 tablet, setting the display output to the second monitor with fixed resolution and the tablet input to the full area leads to reach the edge of the screen when the pen is only halfway across the input; OTB still registers vertical movement, but not horizontal. 

Default drivers work fine, have been uninstalled before installing (and reinstalling) OTB. 

It works fine if the input area is set to be significantly smaller than the actual area. 

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows |
| OpenTabletDriver Version | 0.5.3.3 |
| Tablet                   | XP-PEN Deco 01 V2 |

pcl5x2008, 2021-11-18:
i have the same problem
here is my String dump
Index 1: UGTABLET
Index 2: 10 inch PenTablet
Index 3: 000000
Index 4: UG902_BPG1002
Index 5: 2020-10-28_realse001
Index 6: 
Index 7: 
Index 8: 
Index 9: 
Index 10: 
Index 11: 
Index 12: 
Index 13: 
Index 14: 
Index 15: 
Index 16: 
Index 17: 
Index 18: 
Index 19: 
Index 20: Default mode
Index 21: X_Energy mode
Index 22: Y_Energy mode
Index 23: P_Energy mode
Index 24: Ferq mode
Index 25: Jitter_Energy mode
Index 26: Jitter mode
Index 27: Smooth mode
Index 28: X_Tilt mode
Index 29: Y_Tilt mode
Index 30: 
Index 31: 
Index 32: 
Index 33: 
Index 34: 
Index 35: 
Index 36: 
Index 37: 
Index 38: 
Index 39: 
Index 40: 
Index 41: 
Index 42: 
Index 43: 
Index 44: 
Index 45: 
Index 46: 
Index 47: 
Index 48: 
Index 49: 
Index 50: 
Index 51: 
Index 52: 
Index 53: 
Index 54: 
Index 55: 
Index 56: 
Index 57: 
Index 58: 
Index 59: 
Index 60: 
Index 61: 
Index 62: 
Index 63: 
Index 64: 
Index 65: 
Index 66: 
Index 67: 
Index 68: 
Index 69: 
Index 70: 
Index 71: 
Index 72: 
Index 73: 
Index 74: 
Index 75: 
Index 76: 
Index 77: 
Index 78: 
Index 79: 
Index 80: 
Index 81: 
Index 82: 
Index 83: 
Index 84: 
Index 85: 
Index 86: 
Index 87: 
Index 88: 
Index 89: 
Index 90: 
Index 91: 
Index 92: 
Index 93: 
Index 94: 
Index 95: 
Index 96: 
Index 97: 
Index 98:                
Index 99: 
Index 100: 왰簆῿Ꮨ
Index 101: 最
Index 102: UGEE Technology Corp.
Index 103: 
Index 104: Ă
Index 105: 
Index 106: 
Index 107: 
Index 108: 
Index 109: d
Index 110: ကᄀ
Index 111: ÿ
Index 112: 
Index 113: 
Index 114: 
Index 115: 
Index 116: 
Index 117: 
Index 118: 
Index 119: 
Index 120: 
Index 121: †††††††
Index 122: 
Index 123: HK On
Index 124: HK Off
Index 125: 
Index 126: 
Index 127: 
Index 128: 
Index 129: 
Index 130: 
Index 131: ЁЀ
Index 132: 
Index 133: 
Index 134: 
Index 135: 
Index 136: 
Index 137: 
Index 138: 
Index 139: 
Index 140: 
Index 141: 
Index 142: 
Index 143: 
Index 144: 
Index 145: 
Index 146: 
Index 147: 
Index 148: 
Index 149: 
Index 150: 
Index 151: 
Index 152: 
Index 153: 
Index 154: 
Index 155: 
Index 156: 
Index 157: 
Index 158: 
Index 159: 
Index 160: 
Index 161: 
Index 162: 
Index 163: 
Index 164: 
Index 165: 
Index 166: 
Index 167: 
Index 168: 20180104
Index 169: 
Index 170: 
Index 171: 
Index 172: 
Index 173: 
Index 174: 
Index 175: 
Index 176: 
Index 177: 
Index 178: 
Index 179: 
Index 180: 
Index 181: 
Index 182: 
Index 183: 
Index 184: 
Index 185: 
Index 186: 
Index 187: 
Index 188: 
Index 189: 
Index 190: 
Index 191: 
Index 192: 
Index 193: 
Index 194: 
Index 195: 
Index 196: 
Index 197: 
Index 198: 
Index 199: 
Index 200: 
Index 201: 
Index 202: 
Index 203: 
Index 204: 
Index 205: 
Index 206: 
Index 207: 
Index 208: 
Index 209: 
Index 210: 
Index 211: 
Index 212: 
Index 213: 
Index 214: 
Index 215: 
Index 216: 
Index 217: 
Index 218: 
Index 219: 
Index 220: 
Index 221: 
Index 222: 
Index 223: 
Index 224: 
Index 225: 
Index 226: 
Index 227: 
Index 228: 
Index 229: 
Index 230: 
Index 231: 
Index 232: 
Index 233: 
Index 234: 
Index 235: 
Index 236: 
Index 237: 
Index 238: 
Index 239: 
Index 240: 
Index 241: 
Index 242: 
Index 243: 
Index 244: 
Index 245: 
Index 246: 
Index 247: 
Index 248: 
Index 249: 
Index 250: 
Index 251: 
Index 252: 
Index 253: 
Index 254: 
Index 255:

gonX, 2021-12-01:
A video example and a diagnostics export would be nice from the author of the thread.

gonX, 2021-12-06:
Closing as abandoned

closed 2021-12-06 by gonX