OpenTabletDriver/OpenTabletDriver #1428: IntuosV1 parser incorrectly reporting pen pressure while hovering

EriDere, 2021-09-11:
## Description

IntuosV1 tablets exhibit an odd flickering behavior on the pressure byte while hovering that the current IntuosV1 parser incorrectly reads as pressure. This issue is unique to the pen side of the tablet pen as the eraser side does not exhibit this flickering behavior (as seen in linked video). Hardware issues can be ruled out as the hardware pen status light on the tablet only turns green when actually putting pressure on the pen. 

## Attachments
[Video demonstration of bug](https://youtu.be/DUs7BDtjDU8)
[Tablet data recording](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7147329/gd-0912-u.tablet-data.txt)

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 10 (20H2)
| OpenTabletDriver Version | v0.6.0 pre-release
| Tablet                   | Wacom GD-0912-U

EriDere, 2021-09-11:
I'm using a very small pressure threshold at the moment and it does solve the issue, but I figured I'd note it here as well since it does seem like unintended behaviour. Could understand this being a wontfix due to it being exactly what the tablet outputs though.

EriDere, 2021-09-11:
Upon some further testing, it seems that the tablet uses this "flickering" pressure byte as a way to indicate that the pen is on the pen side and not the eraser, as the pressure byte is the only value that seems to react differently between the pen and eraser sides of the tablet pen. The current IntuosV1 parser also has no way of differentiating between the pen and eraser sides of the pen despite listing both in the configurable tablet settings which is likely related to not parsing the pressure byte for pen orientation. 

Interestingly, the "4D mouse" for this tablet also seems to use flickering values on each report to indicate a different "tool" for the tablet, but the mouse consistently alternates between 2 states every other report while the pen pressure while hovering seems to be random. The mouse also "flickers" the values of more bytes than the tablet pen.

So, as far as i can tell:
|Tool|Behaviour|
|---|---|
|Pen side of pen|<ul><li>Pressure byte (7th byte) randomly flickers between 01 and 00 while hovering</li><li>All other bytes function as expected</ul></li>|
|Eraser side of pen|<ul><li>All bytes function as expected (pressure byte remains as 00 while hovering)</ul></li>|
|4D Mouse|<ul><li>Near proximity/pen button byte (2nd byte) alternates between EA and E8 each report</li><li>Pressure byte (7th byte) reports A8 or rarely A9 when the mouse is on the tablet for each EA report (unclear what the difference between each is). When moving the analogue slider wheel on the mouse, pressure values are reported on this byte for each E8 report based on how far the wheel has been spun (will gather exact values later as needed).</li><li>X Tilt byte (8th byte) switches to a number of unique values on each "EA" report. It is unclear exactly what each of these different values indicates, but I have recorded 20, 40, 80, C0, E0 and A0 (will add more details if I can determine what causes this value to change). E8 reports always report 00 on the 8th byte.</li><li>Y Tilt byte (9th byte) reports the pressed mouse button on each E8 report. From top left button to bottom right, the values are 01, 02, 04, 20, 10.</li><li>All other bytes seem to function as expected (position and hover distance bytes) </ul></li>

I will also update the title of this issue to reflect the fact that it is more related to an expansion in the featureset of the IntuosV1 parser which will add eraser and possibly mouse support rather than a simple pressure parsing bug like I originally assumed.

EriDere, 2021-10-01:
To update this issue with new info:
-  The flickering between 00 and 01 on my particular pen seems to have stopped after breaking in the pen for a couple weeks. The pen still flickers between 2 and 4 pressure while hovering, but this occasionally stops occurring for a bit. Despite other Intuos1 owners reporting similar issues and the tablet pen status light only turning green when actually putting pressure on the pen, I think it can be assumed that this is a common hardware issue with this model and the tablet firmware knows to filter out such reports. Therefore, this should likely be a wontfix.
- The pen eraser not working properly seems to be due to a larger issue with the IntuosV1 parser that expects an IEraserReport on each tablet report despite the tablet only sending a tool detection report on initial hover entry of the pen. Due to this seemingly being a larger issue with the IntuosV1 parser rather than specifically the Intuos1 tablets, this problem should be opened in another issue.
- #1472 adds a mouse parser for Intuos1 tablets and fixes the previous issues mentioned with the 4D mouse.

gonX, 2021-10-01:
> The pen eraser not working properly seems to be due to a larger issue with the IntuosV1 parser that expects an IEraserReport on each tablet report despite the tablet only sending a tool detection report on initial hover entry of the pen. 
I opened an issue on that (#1473) I agree its current implementation is suboptimal.

gonX, 2021-11-30:
I'm going to close this issue as a hardware issue that we won't have to fix since the pressure threshold slider exists for this very reason. 

Eraser end _should_ be detected on master now, though I forgot which of the many Linux artist mode PR's addresses it. Sadly the parser does not cache eraser reports yet and leaves it up to the output mode to cache `IToolReport`'s for now. This should be addressed in a separate issue.

closed 2021-11-30 by gonX