OpenTabletDriver/OpenTabletDriver #1429: No parser support for tablets with 12 express keys

X9VoiD, 2021-09-11:
## Description
There are several tablets that declare 12 express keys, but as of this issue's writing, no report parsers have support for more than 8 express keys.

The following tablets are affected:

- [ ] Gaomon 1060 Plus
- [ ] Gaomon M106K Pro
- [ ] Huion 1060 Pro
- [ ] Huion H1060P
- [ ] Huion H16
- [ ] Huion HS610
- [ ] Huion New 1060 Plus
- [ ] Huion WH1409 V2
- [ ] Veikk A15 Pro
- [ ] Veikk A15

K4LCIFER, 2021-10-19:
The XP Pen artist pro 12 is technically one as well; It has 8 express keys and a scrollwheel.

jamesbt365, 2022-04-06:
Don't they support 12 now?

jamesbt365, 2022-04-19:
Don't we support up to 16 in some cases now? How many can we check off here?

gonX, 2022-11-07:
I went through the list and checked which parsers are used for each tablet, and confirmed the following tablets are now using parsers that _should_ support all their aux keys but they should probably be seen as untested:

`UCLogicParser`/`UCLogicTiltParser`:

_Supports 13 aux keys_
- Gaomon 1060 Pro _(no such thing as Gaomon 1060 Plus)_
- Gaomon M106K Pro
- Gaomon M10K
- Gaomon M10K Pro
- Huion H1060P
- Huion H1161
- Huion HC16 (you wrote H16)
- Huion HS610
- Huion HS611
- Huion Kamvas 13

`Huion.GianoReportParser`:

_Uses same aux parser as `UCLogicParser`, so supports 13 aux keys_
- Gaomon PD1560 _(you wrote PD1650 which doesn't exist)_
- Huion Kamvas 16
- Huion New 1060 Plus _(both variants)_
- Huion WH1409 _(both variants)_
- 

`VeikkReportParser`:
- Veikk A15 Pro **but only in some cases:** (inputreportlength 9 = no, inputreportlength 13 = yes)
- Veikk A15 V2 **but not V1**

`Wacom.IntuosV1Parser`

*This parser will in some cases (depending on bytes sent by the tablet) use `Intuos4AuxReport`, so these I would consider much less likely to work since we have no saved data on what these tablets should be reporting, however I would assume that if one of them works the rest also work:*

- Wacom PTK-540WL
- Wacom PTK-640
- Wacom PTK-840
- Wacom PTK-1240
 
---

I could not find "Huion 1060 Pro", we do however have Huion H610 Pro which should work on everything aside from the H610 Pro V3 which as of this writing (d50c1324488f09d) does not have auxiliary device identifiers in its configuration

So just to reiterate, some devices are not complete yet, most notably:
- Veikk A15 Pro in some cases
- Veikk A15 V2
- H610 Pro V3

I did not check any other tablet configurations other than what was listed in this post.

gonX, 2025-01-26:
I'm going to assume this is fixed almost globally, and tablets that are still affected by this should have their own issue opened.

closed 2025-01-26 by gonX