OpenTabletDriver/OpenTabletDriver #1432: Support Request: Gaomon 1060 Pro wasn't detected as expected

pe200012, 2021-09-12:
# Support Request

I bought a new Gaomon 1060 Pro just yesterday and installed opentabletdriver using AUR trying to make it work under Linux. 

However, the GUI showed `no tablet detected`. `lsusb` returned the vendor info with the name being `Huion Tablet` which was exactly the same as Gaomon 1060 Pro's(`256c:0064`). 

The GUI also crashed when I was trying to generate configuration manually.

## What I've already tried

- uninstall `DIGIMEND` completely
- reinstall opentabletdriver
- uninstall wacom driver

didn't work :(.

Notice that the tablet is working under Windows, and if I force system to recognize this device as wacom ones, it will work with only half the panel.

## Diagnostic Information

[diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7149292/diag.txt)

lsusb:
```
[pe200012@pe200012-room123 ~]$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 256c:0064 HUION Huion Tablet
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
dmesg:
```
[pe200012@pe200012-room123 ~]$ sudo dmesg | grep Huion
[ 2570.519872] usb 1-5: Product: Huion Tablet
[ 2571.161707] hid-generic 0003:256C:0064.0001: hiddev96,hidraw0: USB HID v1.10 Device [HUION Huion Tablet] on usb-0000:00:14.0-5/input0
[ 2571.203187] input: HUION Huion Tablet as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.1/0003:256C:0064.0002/input/input13
[ 2571.203315] hid-generic 0003:256C:0064.0002: input,hidraw1: USB HID v1.10 Device [HUION Huion Tablet] on usb-0000:00:14.0-5/input1
[ 2643.551972] usb 1-5: Product: Huion Tablet
[ 2643.555758] hid-generic 0003:256C:0064.0003: hiddev96,hidraw0: USB HID v1.10 Device [HUION Huion Tablet] on usb-0000:00:14.0-5/input0
[ 2643.595641] input: HUION Huion Tablet as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.1/0003:256C:0064.0004/input/input14
[ 2643.596079] hid-generic 0003:256C:0064.0004: input,hidraw1: USB HID v1.10 Device [HUION Huion Tablet] on usb-0000:00:14.0-5/input1
[ 6223.779749] usb 1-5: Product: Huion Tablet
[ 6223.783227] hid-generic 0003:256C:0064.0005: hiddev96,hidraw0: USB HID v1.10 Device [HUION Huion Tablet] on usb-0000:00:14.0-5/input0
[ 6223.784534] input: HUION Huion Tablet as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.1/0003:256C:0064.0006/input/input15
[ 6223.784721] hid-generic 0003:256C:0064.0006: input,hidraw1: USB HID v1.10 Device [HUION Huion Tablet] on usb-0000:00:14.0-5/input1
[12240.180066] input: HUION Huion Tablet as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.1/0003:256C:0064.0007/input/input16
[12240.180566] hid-generic 0003:256C:0064.0007: input,hidraw0: USB HID v1.10 Device [HUION Huion Tablet] on usb-0000:00:14.0-5/input1
[12240.200628] hid-generic 0003:256C:0064.0008: hiddev96,hidraw1: USB HID v1.10 Device [HUION Huion Tablet] on usb-0000:00:14.0-5/input0
[13588.637879] usb 1-5: Product: Huion Tablet
[13588.661088] hid-generic 0003:256C:0064.0009: hiddev96,hidraw0: USB HID v1.10 Device [HUION Huion Tablet] on usb-0000:00:14.0-5/input0
[13588.702303] input: HUION Huion Tablet as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.1/0003:256C:0064.000A/input/input17
[13588.702792] hid-generic 0003:256C:0064.000A: input,hidraw1: USB HID v1.10 Device [HUION Huion Tablet] on usb-0000:00:14.0-5/input1
```

shugen002, 2021-09-22:
look like you are using another hardware version of Gaomon 1060 Pro . ~~My Gaomon 1060 Pro is `256C:006D`. __You can try edit the configuration in `Tablet->Open Configuration Editor`.~~
~~For your tablet , you may need to change `Gaomon 1060 Pro -> Digitizer identifiers -> Digitizer identifiers -> Product ID` from `109` (006D) to `100`(0064).~~
After some search , looks like you are using Gaomon M6 instead of Gaomon 1060 Pro ?

pe200012, 2021-10-01:
@shugen002 Wait, are there multiple versions of 1060 pro? My tablet is identified as 1060 pro under Windows using the official driver, so I didn't even think of this possibility.

X9VoiD, 2021-10-01:
There are many variations of "1060 Pro", Huion, UC Logic, Gaomon, even variations with the Huion variant itself.

pe200012, 2021-10-03:
Change product id in OTD's profile from `006D` to `0064` as @shugen002 told fixed the problem.

closed 2021-10-03 by pe200012

shugen002, 2021-10-03:
> Change product id in OTD's profile from `006D` to `0064` as @shugen002 told fixed the problem.

Although i don't think that a good idea , still glad to see it solved.