OpenTabletDriver/OpenTabletDriver #1436: Support Request: OTD v0.5.3.3 wont detect veikk s640 v2

mozurino, 2021-09-14:
the title says it all

jamesbt365, 2021-09-14:
you need to use the pre release, im unsure where to get it on github so you can either look for it or join the opentabletdriver discord server and grab it from the #pre-releases channel

Mrcubix, 2021-09-14:
Get a refund if you can, then buy an S620 OR  a used 470 471 OR new 472 

OR

Build to use: https://cdn.discordapp.com/attachments/615939905566015493/884578185390456862/OpenTabletDriver.win-x64.zip

:warning: DO NOT INSTALL PLUGINS FROM THE LIST AVAILABLE IN PLUGIN MANAGER AS IT WILL CAUSE OTD TO CRASH DUE TO IMCOMPATIBILITY AND SOME OTHER BUGS :warning: 

Links of the only plugins that work in this current Unstable dev build (Drag and drop these as they are in the plugin manager):

KUUUBE's CHATTER ETXERMINATOR (anti-chatter): https://cdn.discordapp.com/attachments/628651971267788800/884592400402632744/Kuuubes_CHATTER_EXTERMINATOR.dll

Reconstrcuctor (Anti hardware-smoothing):
https://github.com/X9VoiD/VoiDPlugins/suites/3336736897/artifacts/78275611

If you play the dog sh*t mobile copy pasta with kernel toaster anti-cheat using your tablet:

Requirement: https://github.com/X9VoiD/vmulti-bin/releases/latest
Vmulti: https://github.com/X9VoiD/VoiDPlugins/suites/3336736897/artifacts/78275614

closed 2021-09-15 by InfinityGhost