OpenTabletDriver/OpenTabletDriver #1445: Add support for Huion RTP-700

Sashamomo, 2021-09-18:
# Tablet Information

| Name  Huion RTP0700 | Value |
| --------------- | ----- |
| Manufacturer  Huion  |
| Product Name RTP-700  |
| Width 279.4 |
| Height 174.6 |
| Pressure Levels 8192 |
| LPI   5080 |

## Diagnostic Information

[HUION RTP-700.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7189756/HUION.RTP-700.txt)

Mrcubix, 2021-09-18:
That is not an OTD diagnostic file, please provide a proper diagnostic file:

In the interface : Help > Export diagnostocs, then upload it here

Mrcubix, 2021-09-18:
Uninstall huion drivers and run the [cleanup](https://github.com/X9VoiD/WinUSBCleanup/releases/tag/v1.0), then replug your tablet and export a diagnostic file again

Sashamomo, 2021-09-18:
[diag (1).txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7189857/diag.1.txt)

Sashamomo, 2021-09-18:
[diag_2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7189858/diag_2.txt)

Mrcubix, 2021-09-18:
alright we can now get started with getting this tablet compatible, lemme work on the config real quick.

Sashamomo, 2021-09-18:
Ok, thanks

Mrcubix, 2021-09-18:
can you go in `Tablets > device string reader` and type the following values in each boxes there:

- 9580
- 109
- 201
- click on send request and send the obtained string, also click on dump all, save the file and upload it here

Sashamomo, 2021-09-18:
HUION_T19k_210225
[String dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7189871/String.dump.txt)

Mrcubix, 2021-09-18:
[Huion RTP-700.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7189874/Huion.RTP-700.zip)
extract the content of this zip file in `Configurations\Huion`

Sashamomo, 2021-09-18:
It detect but it didn't work

Mrcubix, 2021-09-18:
> It detect but it didn't work

Define `didn't work`

Mrcubix, 2021-09-18:
when that's done, go to `Tablets > detect tablet` and then `Tablets > tablet debugger`,
once you're there, you'll need to move your pen around, values will appear on screen.
Confirm that it happen so we can continue.

Sashamomo, 2021-09-18:
Debagger is ok. The absolute mode works fine, the relative mode doesn't work.

Mrcubix, 2021-09-18:
wdym by relative mode doesn't work?

Mrcubix, 2021-09-18:
also since you mention that values appear in the debugger let's finish this configuration and continue after:

In the debugger:
if you look on the right, you'll be able to see the value named `Position`.
I need you to move to the bottom right of the tablet very slowly and get the maximum value you can.

Sashamomo, 2021-09-18:
I relative mode when moving a pan the cursor don't move.

Sashamomo, 2021-09-18:
[image]

Mrcubix, 2021-09-18:
[Huion RTP-700.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7189944/Huion.RTP-700.zip)
replace the previous config with this one and then `Tablets > Detect Tablet`

Sashamomo, 2021-09-18:
Now all works fine

Mrcubix, 2021-09-18:
check all of the tablet's features just to be sure

Sashamomo, 2021-09-18:
Both Multi modes don't work. Auxiliary buttons don't work too

doubledge-spec, 2021-09-18:
for vmulti you need another thing
the aux is a config change probably

Mrcubix, 2021-09-18:
[Huion RTP-700.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7190037/Huion.RTP-700.zip)
try this config for aux, as for VmultiMode plugin, you'll need the [requirement](https://github.com/X9VoiD/vmulti-bin/releases/latest) to make it work

Sashamomo, 2021-09-18:
VmultiMode works. Auxiliary buttons don't work.

Mrcubix, 2021-09-18:
in tablet debugger: does any bytes change when you push any of the buttons

Sashamomo, 2021-09-18:
Nothing change

Mrcubix, 2021-09-18:
then the endpoint isn't reporting these aux buttons.

- https://desowin.org/usbpcap/thankyou.html?file=1.5.4.0/USBPcapSetup-1.5.4.0.exe Download and install this tool

- When done, head to C:\Program Files\USBPcap in the explorer (you can probably copy paste that path in the path bar)

- Once you're there, start USBPcapCMD.exe

- A console will appear, answer n at the first prompt and reboot your pc

- Head back to C:\Program Files\USBPcap,
- Start USBPcapCMD.exe,

- screenshot the console.

jkcdarunday, 2021-12-30:
On Linux, everything seems to be working (including buttons) with the latest config except for tilt support. I don't see tilt being reported when running `libinput debug-gui` as well as discussed in #1231 

When using the Tablet Debugger, I don't see anything related to tilt in the Tablet Report section. However, I do notice the last two hex values in the Raw Tablet Data section changing according to how I tilt my pent. Here's a recording of how the values change as I tilt my pen up - down - left - then right. At the end, I rotate my pen without tilting showing minimal change in the last two hex values implying that they indeed likely represent pen tilt values.
https://user-images.githubusercontent.com/4564810/147727011-bc95ecf2-2712-41a1-8842-40ad04a98b5d.mp4

gonX, 2021-12-30:
@Mrcubix when the config eventually gets into prerelease the new `UCLogicTiltParser` should address the above issue

jamesbt365, 2021-12-31:
Are we still waiting for some windows information?

jkcdarunday, 2021-12-31:
Uploading here the config that worked for me with the latest master build (I believe it's being called 0.6) on Linux:
[RTP-700.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7795130/RTP-700.zip)

Mrcubix, 2021-12-31:
Yeah but it's not complete, Aux was apparently not working so a USB capture is still required on windows just to get the necessary info

jkcdarunday, 2021-12-31:
For the tilt, I tried switching the `ReportParser` to `OpenTabletDriver.Configurations.Parsers.UCLogic.UCLogicTiltReportParser` after building the latest `configs` branch but the GUI crashes so I cherry picked the tilt report parser changes into my own fork of the master branch which seems to have worked and gave me tilt support.

I did notice though that one of the tilt axes were inverted when I used Krita and MyPaint on Linux so I added a change that inverts one of the axes in my fork. Not sure if this is due to Krita doing some magic, a change that I didn't get from the configs branch, an issue specific to RTP-700, or that's just how UC Logic values work.

Here are the changes that worked for me in case it's useful to @gonX:
https://github.com/jkcdarunday/OpenTabletDriver/commit/314b9590d762ab0026ad08a7c74de34198b4172f

Mrcubix, 2021-12-31:
you could just PR the change, just make sure you're based on the `configs` branch and are targeting the `config` branch in the PR

Sashamomo, 2022-04-02:
[image]

Sashamomo, 2022-04-02:
> then the endpoint isn't reporting these aux buttons.
>
> * https://desowin.org/usbpcap/thankyou.html?file=1.5.4.0/USBPcapSetup-1.5.4.0.exe Download and install this tool
> * When done, head to C:\Program [...]

[image]

Sashamomo, 2022-04-27:
Hello. Can we continue working on the configuration?

Janus-M, 2024-07-11:
Here it is new version. 
[RTP-700 (1).json](https://github.com/user-attachments/files/16172214/RTP-700.1.json)

PR #3366 (2024-07-31): Add support for Huion RTP-700
Fixes #1445.

Diagnostic: https://discord.com/channels/615607687467761684/615611007951306863/1260668502084288584
Verification: https://discord.com/channels/615607687467761684/615611007951306863/1260673849255661639
closed 2024-07-31 by InfinityGhost